mrrwa/NmraDcc #3: Need a hand

jhdulaney, 2016-06-06:
I've tried running the various examples with no luck at all.  I don't have an oscilloscope, so I can't be quite sure that I'm actually getting a DCC signal to the Arduino.  Any pointers?

kiwi64ajs, 2016-06-06:
Hmmm...
Off the top of my head:

1) what is your input circuit?

2) Does it need a pull-up resistor - you can enable that on the call to Dcc.init()

3) If it's nothing simple or obvious then you need to figure out if you're actually receiving the DCC signal on the INTx pin. 

To debug this I'd:
- Write a short bit of code that has an array of long integers (say 50 or 100 slots) to hold the value returned from the micros() function and an index pointer to point to the next slot to deposit the values from micros() into.
- Then write an INTx interrupt handler that monitors the DCC Input signal and triggers on a state change and capture the value returned by the micros() function and store it in the array.
- Once the array is full, print out the contents of the array and subtract consecutive values to get the delta timings of the transitions. You're looking for a waveform that has periods of about 55us or 110us.

This is the approach I took to develop this stuff in the first place - I didn't have a scope or logic analyser until years after - I simply used print statements (output a single char from the interrupt handler at high baud rate) and captures of timing values like I just described - you're in good company...

HTH

Alex Shepherd

jhdulaney, 2016-06-06:
This is the circuit I am using for input:

https://rudysmodelrailway.files.wordpress.com/2014/07/schematic1.png

I put this together just to see if I could tell if it was changing states or not:

byte state = 0;
byte last_state = 0;

void setup() {
  Serial.begin(115200);
  pinMode(2, INPUT);
  

}

void loop() {
  state = digitalRead(2);
  if (state != last_state) {
    if (state == HIGH) {
      Serial.println("High");
    }
    else {
      Serial.println("Low");
    }
  }
  last_state = state;

}

It does not appear to do so.

kiwi64ajs, 2016-06-07:
Hmmm...
I'd connect a multimeter set on AC between GND and Arduino pin 2 and see if any signal is getting to the input pin

Regards

Alex Shepherd

devel-bobm, 2016-06-07:
> -----Original Message-----
> FpsImwUD26zfquPAfeLGks5qJJSKgaJpZM4IvPFy> .

John,

Which Arduino board "style" are you using?  This might make a difference since the different boards can use different Atmel microcontrollers with varying capabilities!

Are you connecting to a pin numbered "2" on the board itself, or to the "Arduino pin" "D2"?  There can be a difference between those two concepts...

Have you shown proper serial communication between the Arduino and the computer, without using the DCC library?

I remember purchasing (25+ years ago) some 1N4148 diodes from Radio Shack only to find that the "band" printed on the device was not where I expected it to be - it was on the anode end of the diode body rather than the cathode end, where the band is typically printed.  I had to reverse the 1n4148 diodes in the circuit to get it to work.  I don't remember if this was properly documented on the package (i.e. stupidity on my part) or if it was a "mistake" by the manufacturer and/or incorrect info on the R/S blister-pack.

Regards,
Billybob

kiwi64ajs, 2016-07-23:
As there are no more comments I presume this is resolved and am closing the issue

closed 2016-07-23 by kiwi64ajs