mrrwa/NmraDcc #6: Address Incorrect?

tomas-kucera, 2016-08-04:
Hi there,

the library works awesome and I love it. I managed to build my turnout controller using it.

The only issue I am facing is that the address seem to be interpreted incorrectly.

If I use my Roco 10764 Base Station with multiMouse and send accessory enable / disable to an address. Then by sniffing within notifyDccAccState method by printing out the Addr (as DEC), I am getting values lower by 4 ... for example sending the command to Addr 5 prints out Addr=1, etc.

I can certainly fix it by adding 4 to it, but I would rather have this sorted out at the library level.

Any idea?

Thanks in advance!

bobjacobsen, 2016-08-04:
Unfortunately, the [NMRA specification for accessory decoder addresses](http://www.nmra.org/sites/default/files/s-9.2.1_2012_07.pdf) (starting at 420) is a mess.  Attempts to improve that situation failed due to NMRA politics.  So now we have at least three different implementations in command stations that, for the same number at the human interface, send different values in the DCC packet.

One of the differences in implementations is whether 1 is the lowest value of the AAAAAAAAA field in the standard or whether to use 0 as the lowest value.

This is not really the fault of the library. The library is implementing the by-far most popular coding mechanism; it's just that the Roco implementation is using a different one.

tomas-kucera, 2016-08-04:
OK, thanks for the clarification!

closed 2016-08-04 by tomas-kucera

kiwi64ajs, 2016-08-04:
Hi Guys,

Yeah this has come up before but I haven’t added any new features to accomodate the addressing ambiguity but here us a comment I have made previously which may or may not be useful:

> Looking at the Roco website FAQ I see this comment:
> 
> 	http://www.z21.eu/en/FAQ-Support/FAQ <http://www.z21.eu/en/FAQ-Support/FAQ> question: “Why are the turnout addresses of some turnout decoders shifted by 4?”
> 
> To ease the transition or the use of external decoders for customers, we offer the option "DCC turnout address displacement +4" in the Z21 Maintenance Tool. If this option is set, the "translation" from the turnout address to the module address and to the compatible port number of the external control system will be made in the Z21. Thus, for example, the turnouts and signals can be continued to be used with the previously used turnout addresses.
> Finally we would like to mention that this problem only occurs in relation to the DCC. With the Motorola turnout decoders the turnout addresses and track signal generation is clear.

So one work-around is to use the "DCC turnout address displacement +4” solution described above if the Command Station supports it. 

Another option could be to provide a member function to set the First Address Offset, but default it to 1.

Yet another work-around might be to add a CV that defines the First Address Offset and default it to 1 to allow compatibility with the existing library, but allow the CV value for individual decoders to have different values for the first address where needed.

Not sure which might give the best outcome all-round - comments?

Regards

Alex Shepherd

MicroBahner, 2016-08-11:
Well, I'm not sure wether this is worth to change the library. If you know about the problem, it is easy to add the offset of 4 in the sketch ( as thomas mentioned already ). That is'nt more complex than calling a memberfunction or setting a CV.
And if you don't know about the problem ... you will not be aware that you have to call a function or set a CV.
Maybe it's worth a note in the description of the notifyDccAccState() function.

Franz-Peter