mrrwa/NmraDcc #7: SMB Examples drop packets

MicroBahner, 2016-08-15:
Hi,
I found out, that the SMB Examples lose packets. The reason is that the Dcc.process() function isn't called often enough. 
The Dcc.process() function must be called at least once for every received packet. A DCC accessory packet may be around 6ms long ( including the preamble ). Therefore the Dcc.process() must be called more than once every 6ms.  Because of the delay(8) in the loop of the examples this is not possible.

Normally this will not be apparent, because the central station repeats the packets several times.
But I think especially the examples should be programmed to not lose packets under normal conditions.

kiwi64ajs, 2016-08-15:
Good catch!
I've CC'd Geoff

Regards

Alex Shepherd

kiwi64ajs, 2016-11-27:
I've just merged changes from Geoff that should address this issue

closed 2016-11-27 by kiwi64ajs