mrrwa/NmraDcc #9: [Question] DCCFunc, DDCSpeed and DCCTurnout at the same time

littleyoda, 2016-08-27:
For my project, I want to receive DDCFunc, DDCSpeed and DDCTurnout.

Depending on the Parameter for the Constructor, I  got DCCFunc and DDCSpeed or just DDCTurnout.
But I wasn't able to receive all three information. Is there an way to accomplish this?

kiwi64ajs, 2016-08-28:
Hi Sven,

Unfortunately the library has a number of global resources (CV Storage for one) and really assumed to be either a Mobile or Stationary Decider - not both at the same time. The fundamental issue is the single copy of the CV Storage and really only allowing for one call for CV29 which really determines the overall behaviour of the library. Also the call-back functions are global.

To support what you’re wanting to do we require some subtle and non-trivial changes to the core and would require separating out the codebase into distinct modules so you could rearrange things.

For instance if we separated the DCC Packet receive logic out to a module (DCC-COMMS) and the CV storage logic out to another module (CV-STORE) from the rest or the library (DCC-CORE) then we could allow multiple instances of the DCC-CORE, each with their own CV_STORE objects, to be fed DCC Packets from a single DCC-COMMS module. We’d have to chang the whole call-back functions as well so it would not be trivial or easy to do what you’re asking without a lot of effort.

Regards

Alex Shepherd

p: +64-7-829 4441
m: +64-21-777764
e: kiwi64ajs@gmail.com

bobjacobsen, 2016-08-28:
There's a more fundamental problem than that:  The NMRA in its infinite wisdom defined the packet formats such that you _can't_ properly decode them unless you know that you're decoding for an accessory decoder or a multi-function (locomotive) decoder.  

Yes, you can look at the address range to see which is it, but the decoding is still context-specific.

railnerd, 2016-08-28:
Have you guys looked at my friend Kevin Snow’s DCC decoding library?   He implemented a full protocol sniffer.

https://github.com/MynaBay/DCC_Decoder

-Dave

littleyoda, 2016-08-28:
I understand what there is a CV-Problem in same cases.

But for my project, I need something similar to a sniffing/monitoring tool. No need for CV programming or for filtering of the commands by decoder addresses.

@railnerd: I testet MynaBay/DCC_Decoder. But the results are not so good. Only 10% of the commands are really show on my esp8266. All other commands are missing.

This morning, I found a solution for my problem:
In processMultiFunctionMessage I removed the code between "uint8_t CmdMasked = Cmd & 0b11100000;"  and "switch (CmdMasked) {". Faw away from a elegant solution, but I seems to work.
I received now all three kinds of information.

A Sniffer or Monitoring Modus would be great.

MicroBahner, 2016-09-01:
Hi Sven,
maybe the notifyDccMessage()  funktion ist a solution for you? You get a pointer to the last received packet. You see every packet on the line, but without any interpretation.

Regards
Franz-Peter

kiwi64ajs, 2016-11-13:
The library fundamentally can only be one of either a Mobile or Stationary decode at the same time as there are a bunch of assumptions around CV29 handling that hake it difficult to dynamically make it respond to both. I'm not prepared to change this right now

closed 2016-11-13 by kiwi64ajs