mrrwa/NmraDcc #10: use of esp8266: SPI_FLASH_SEC_SIZE not declared in this scope

digi-thomas2003, 2016-11-13:
Hello,

if I try to compile a sketch (e.g. one of the examples delivered with the lib) using the current release of the NmraDcc lib I get the following error message in the Arduino IDE:

_Arduino: 1.6.12 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 921600, 4M (3M SPIFFS)"

In file included from C:\Users\x\Documents\Arduino\libraries\NmraDcc\NmraDcc.cpp:33:0:

C:\Users\x\Documents\Arduino\libraries\NmraDcc\NmraDcc.cpp: In function 'uint8_t validCV(uint16_t, uint8_t)':

C:\Users\x\Documents\Arduino\libraries\NmraDcc\NmraDcc.h:94:21: error: 'SPI_FLASH_SEC_SIZE' was not declared in this scope

   #define MAXCV     SPI_FLASH_SEC_SIZE

                     ^

C:\Users\x\Documents\Arduino\libraries\NmraDcc\NmraDcc.cpp:494:12: note: in expansion of macro 'MAXCV'

   if( CV > MAXCV )

            ^

C:\Users\x\Documents\Arduino\libraries\NmraDcc\NmraDcc.cpp: In member function 'void NmraDcc::init(uint8_t, uint8_t, uint8_t, uint8_t)':

C:\Users\x\Documents\Arduino\libraries\NmraDcc\NmraDcc.h:94:21: error: 'SPI_FLASH_SEC_SIZE' was not declared in this scope

   #define MAXCV     SPI_FLASH_SEC_SIZE

                     ^

C:\Users\x\Documents\Arduino\libraries\NmraDcc\NmraDcc.cpp:1015:18: note: in expansion of macro 'MAXCV'

     EEPROM.begin(MAXCV);

                  ^

exit status 1_

Should this be defined somewhere in the nmradcc.cpp or nmradcc.h file? Or am I missing something else?

Thanks for any help

devel-bobm, 2016-11-13:
It appears that the NmraDcc library is using a board-specific or processor-specific definition.  I would guess that the appropriate SPI library for the ESP8266 would be responsible for defining this.
@ kiwi64ajs, can you provide any guidance?

kiwi64ajs, 2016-11-13:
I forgot to re-tag the 1.4.0 release so the Arduino Library Manager update mechanism didn't see the change. I've done this a short time ago so that change should come through soon.

digi-thomas2003, 2016-11-15:
Hi Alex,

sorry I have to eopen this issue.
After updating the lib to release 1.4.0 as you stated above this error message is still occuring.
If I replace the SPI_FLASH_SEC_SIZE with the value 4096 the compilation runs through.
So I think the lib does not find the declaration of this name.
Can you please check this?
Thank you!
Regards
Thomas

kiwi64ajs, 2016-11-15:
Hi Thomas,
> sorry I have to eopen this issue.
> After updating the lib to release 1.4.0 as you stated above this error message is still occuring.
> If I replace the SPI_FLASH_SEC_SIZE with the value 4096 the [...]
Hmmm.. that’s odd, something must have changed i the ESP8266 core.

Anyway I’ve tweaked a few things and committed version 1.4.1 check in a few hours to see if this change becomes available in the Arduino Library manager.

Version 1.4.0 should have disappeared... 

Regards

Alex Shepherd

p: +64-7-829 4441
m: +64-21-777764
e: kiwi64ajs@gmail.com

digi-thomas2003, 2016-11-16:
Hello Alex,

thank you. Now the compiling works without any trouble!
You can close this issue now as far as I am concerned.

Best regards
Thomas

closed 2016-11-16 by kiwi64ajs