mrrwa/NmraDcc #22: ESP32 support

tanner87661, 2019-02-15:
After successfully running the Accessory test program that is part of the library I started using the code in a larger program. It started crashing as soon as the DCC signal was present, which pointed my look closer into the ISR.

In the process I think I have discovered some problems with the ISR:
a) for ESP32, it should be compiled with the IRAM_ATTR attribute like this: void IRAM_ATTR ExternalInterruptHandler(void)
For more information see here: https://techtutorialsx.com/2017/09/30/esp32-arduino-external-interrupts/
and here: https://github.com/espressif/arduino-esp32/issues/489

b) I also added the MUX line: portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED; and secured the access to the shard data in the ISR and the process routing:
portENTER_CRITICAL_ISR(&mux); DccRx.PacketCopy = DccRx.PacketBuf ; DccRx.DataReady = 1 ; portEXIT_CRITICAL_ISR(&mux);

Unfortunately, while all this is good, it was still crashing as soon as my program tried to access the file system (SPIFFS), eg. for loading config files or a serving a web page.

c) It turns out the real problem is the reattachment of interrupts within the IRS. As soon as I commented those lines out, the program stopped crashing, but unfortunately, it would no longer decode the DCC signals. So I found a workaround and defined a variable ISRWatch which holds the type of interrupt to look for, either falling, raising, or change. Then I replaced all attach interrupt functions by just changing the content of the ISRWatch variable and changed the attacheinterrupt call in the init function to CHANGE. Finally, I added a filter to the beginning of the ISR where I would check the type of interrupt and either return or proceed:

void IRAM_ATTR ExternalInterruptHandler(void)
{
switch (ISRWatch)
{
case CHANGE: break; 
case RISING: if (digitalRead(DccProcState.ExtIntPinNum)) break; 
case FALLING: if (digitalRead(DccProcState.ExtIntPinNum)) return; break; 
}

In this quick fix I did not care about support for other processors etc. So maybe someone with good knowledge of the entire could make the proper adjustments so it works with all processors.

kiwi64ajs, 2019-02-15:
> and here: espressif/arduino-esp32#489

Hi Hans

Can you post your modified code or email it to me please so I can see it in context?

Alex

tanner87661, 2019-02-15:
Hi Alex

 

See attached. Changes are in line 250 and then lines 350-10xx of nmradcc.cpp

I think there are no changes in the header file

 

Best wishes

Hans

 

// Model Railroading with Arduino - NmraDcc.h 
//
// Copyright (c) 2008 - 2018 Alex Shepherd
//
// This source file is subject of the GNU general public license 2,
// that is available at the world-wide-web at
// http://www.gnu.org/licenses/gpl.txt
// 
//------------------------------------------------------------------------
//
// file:      NmraDcc.h
// author:    Alex Shepherd
// webpage:   http://mrrwa.org/
// history:   2008-03-20 Initial Version
//            2011-06-26 Migrated into Arduino library from OpenDCC codebase
//            2014 Added getAddr to NmraDcc  Geoff Bunza
//            2015-11-06 Martin Pischky (martin@pischky.de):
//                       Experimental Version to support 14 speed steps
//                       and new signature of notifyDccSpeed and notifyDccFunc
//            2017-11-29 Ken West (kgw4449@gmail.com):
//                       Added method and callback headers.
//
//------------------------------------------------------------------------
//
// purpose:   Provide a simplified interface to decode NMRA DCC packets
//			  and build DCC MutliFunction and Stationary Decoders
//
//------------------------------------------------------------------------

// Uncomment the following Line to Enable Service Mode CV Programming
#define NMRA_DCC_PROCESS_SERVICEMODE

// Uncomment the following line to Enable MultiFunction Decoder Operations
#define NMRA_DCC_PROCESS_MULTIFUNCTION

// Uncomment the following line to Enable 14 Speed Step Support
//#define NMRA_DCC_ENABLE_14_SPEED_STEP_MODE

#if defined(ARDUINO) && ARDUINO >= 100
#include "Arduino.h"
#else
#include "WProgram.h"
#endif

#include "EEPROM.h"

#ifndef NMRADCC_IS_IN
#define NMRADCC_IS_IN

#define NMRADCC_VERSION     200     // Version 2.0.0

#define MAX_DCC_MESSAGE_LEN 6    // including XOR-Byte

typedef struct
[... 1771 lines cut ...]
          DB_PRINT("eDP: OAddr:%d", OutputAddress);
          
          if( DccProcState.inAccDecDCCAddrNextReceivedMode)
          {
          	if( DccProcState.Flags & FLAGS_OUTPUT_ADDRESS_MODE )
          	{
              DB_PRINT("eDP: Set OAddr:%d", OutputAddress);
			  //uint16_t storedOutputAddress = OutputAddress + 1; // The value stored in CV1 & 9 for Output Addressing Mode is + 1
          	  writeCV(CV_ACCESSORY_DECODER_ADDRESS_LSB, (uint8_t)(OutputAddress % 256));
          	  writeCV(CV_ACCESSORY_DECODER_ADDRESS_MSB, (uint8_t)(OutputAddress / 256));
          	  
          	  if( notifyDccAccOutputAddrSet )
          	  	notifyDccAccOutputAddrSet(OutputAddress);
          	} 
          	else
          	{
              DB_PRINT("eDP: Set BAddr:%d", BoardAddress);
          	  writeCV(CV_ACCESSORY_DECODER_ADDRESS_LSB, (uint8_t)(BoardAddress % 64));
          	  writeCV(CV_ACCESSORY_DECODER_ADDRESS_MSB, (uint8_t)(BoardAddress / 64));
          	  
          	  if( notifyDccAccBoardAddrSet )
          	  	notifyDccAccBoardAddrSet(BoardAddress);
          	}
          	
          	DccProcState.inAccD

kiwi64ajs, 2019-02-16:
> // 2015-11-06 Martin Pischky (martin@pischky.de <mailto:martin@pischky.de>):
> // 2017-11-29 Ken West (kgw4449@gmail.com <mailto:kgw4449@gmail.com>):
> // 2018-12-17 added ESP32 support by Trusty (thierry@lapajaparis.net <mailto:thierry@lapajaparis.net>)

Hi Hans,

I’m having trouble to get this to work on the AVR so maybe you’ve missed something so can you email me the actual NmraDcc.h and NmraDcc.cpp files as attachments please?

I seem to get the first packet and then it freezes.

Alex

tanner87661, 2019-02-16:
Hi Alex

 

I just double checked. What I sent you earlier are my current files. Here are the details of my changes:

 

Line 250: Added new variable ISRWatch

Line 305: Added PortMUX_Type

Line 307: Added IRAM_ATTR attribute

Line 309-313: added switch statement to see what type of interrupt we are dealing with and what to do

Line 413: eliminated Interrupt reattachment and just changed interrupt tyupe in ISRWatch

Same in Line 453 and 485

Line 524 and 527: added critical section for access to shared variable 

Line 1352: changed attachinterrupt parameter from RISING to CHANGE

Line 1463 and 1466: added critical section for access to shared variables

 

I would think that for the AVR platform some of these changes will not work, so you need to add conditional compilation, probably even go back to changing the interrupts

That’s it.

Hans

kiwi64ajs, 2019-02-16:
Hi Guys,

I've hand-merged the ESP32 IRAM_ATTR changes and replaced the detach/attachInterrupts code and added the portENTER/EXIT_CRITICAL sections as well. I've pushed the changes on the  ESP32-IRAM_ATTR  branch here: https://github.com/mrrwa/NmraDcc/tree/ESP32-IRAM_ATTR

I've confirmed that the code still works on an AVR, but it would be helpful if some of you were able to confirm that it works on the ESP32 and STM32F1 platforms as well.

Alex

MicroBahner, 2019-02-16:
Hi Alex, Hi Hans
I saw the changes this morning and already had a look at them. I don't think it's a good idea to have these ESP32 related changes of Hans for all processors. We get much more Interrupts than necessary. This may not be a problem with ESP32 because it's considerably faster then AVR, but we should not do the changes generally. I think It is better and 'safer' to make them ESP32 specific and leave the code for the other processors unchanged. 
I try to make a proposal in this sense the next days.

Regards
Franz-Peter

MicroBahner, 2019-02-16:
Hi Hans,
when looking over the code I had another idea. During my test for the STM32F1 processors I had a similar problem with the reattachment of the interrupts. To get it working I had to explicitly detach it bevor reattaching again. Such as here in the code:
`#if defined ( __STM32F1__ ) `
`     detachInterrupt( DccProcState.ExtIntNum );`
`#endif`
`attachInterrupt( DccProcState.ExtIntNum, ExternalInterruptHandler, CHANGE);`

Did you check whether this could be a solution for ESP32 too? If this works it would be better than creating unnecessary interrupts. And the code would be more consistent across the different processors.

Franz-Peter

tanner87661, 2019-02-16:
Hi Franz-Peter

 

Yes, that was the first thing to try. It had no effect. The general rule for ISRs on the ESP32 is that they (as well as all other functions called from the ISR) must be in the RAM, not in the flash, or you will see crashes when using other flash intensive stuff like for example SPIFFS file server (there is actually plenty of documentation on this problem). And attach/detach interrupt is not a function in RAM and takes some of time, therefore the problems.

Personally I am also not sure if the solution with changing the interrupt trigger is really good on the AVR platform. Someone should investigate how many cycles it cost to reattach the interrupt compared to returning from an unused interrupt immediately.

Hans

MicroBahner, 2019-02-16:
Hi Hans,
ok, so we have to stick to your solution.

> Personally I am also not sure if the solution with changing the interrupt trigger is really good on the AVR platform. Someone should investigate how many cycles it cost to reattach the interrupt compared to returning from an unused interrupt immediately.

Reattaching the interrupt occurs not in every interrupt, but only 2 times during a whole DCC packet. Unnecessary interrupts occur in nearly every bit, because the change irq is only used during a short period after detecting the preamble and searching for the startbit.

Franz-Peter

kiwi64ajs, 2019-02-16:
I might dig into the Arduino core library and see how hard it might be to just change the edge as opposed to detach/attach. It’d be a hack but...

A potentially better solution on the ESP32 is to use the RMT peripheral but that is a huge change.

Alex

MicroBahner, 2019-02-17:
Hi Alex, Hi Hans,
now I created a branch with the ESP32 specific changes of Hans. If 'ESP32' is defined it should  work as Hans proposed with all his changes. If 'ESP32' is not defined it should create a code just like the current master branch. A first test on my standard-Decoder HW worked. I will do some more testing the next days for AVR and STM32F1. Unfortunately I don't have a ESP32. So I'd like to ask Hans to test this code. It is in  the branch [ESP32-processor-specific](https://github.com/MicroBahner/NmraDcc/tree/ESP32-processor-specific) of my fork of the lib. If the tests are successfull I can create a pull request.

In this solution only the ESP32 will create that many unnecessary Interrupts. I hope they don't matter as much as on AVR.

--------------------------
A complete other solution that we can discuss might be, to avoid changing the interrupt trigger in the ISR and do it outside in the 'normal' sketch. As I understand, that schould work on ESP32 too.
Which trigger - rising or falling - is needed depends on the polarity of the DCC signal. As it is now, this is determined after the preamble when waiting for the startbit of a packet. The CHANGE irq is needed to do this until startbit and polarity are detected. But obviously - if you determined the polarity once it will not really change between every packet. Usually it changes only if you unplug and plug in the other way round. On the other hand - if you are on the false edge, you will never get a correct packet. Either you get a checksum error or you get no packet at all.
So my idea is: In the process method you check every second whether you received at least one correct package within the last second. If so, everything is ok and nothing has to be done. If you received only checksum errors or no packet at all, then change the IRQ trigger to the other edge and wait for the next second. Of course when applying the DCC signal the first time ore changing the polarity it may last 1 or 2 seconds until packets are detected correctly.
What do you think - might that be a solution for all processors?  Or is there an error in my reasoning?

tanner87661, 2019-02-17:
Hi Franz-Peter

 

Thanks for working on this.

 

I will give it a try, probably tomorrow, but from looking over it I think it should work except that line 1395 should be eliminated. If the interrupt is not changing, it does not need to be reattached.

 

Changing the interrupt type from outside might work, I think.

 

Polarity changes can actually be more frequent than what you describe. Think about a decoder that sits behind an auto-reversing section.

 

Also, how does this code react in case of zero-stretched zero bits, when using an analog locomotive?

 

A completely different approach might be to use change type interrupt but only record signal level and last eight bits of micros() into a buffer so that the interrupt returns fast. Then use the process routine (or in the case of RTOS even define  task for this) to do the analysis of the stream and decoding. I remember back in 1997 I fed the DCC stream into a sound card and decoded it from the WAV stream using that approach. At the time I think it was the first PC based DCC decoder running on a 386 under Windows 3.11 I believe it was. I called it DCC Sniffer and even got a cease and desist letter from a company who was marketing a product called Network Sniffer, because they had a TM on the term Sniffer ;-) Tempi passati.

 

Hans

MicroBahner, 2019-02-18:
Hi Hans,

> except that line 1395 should be eliminated.

Really? Line 1395 is the inital attach in the ini-function after starting the sketch. I think you cannot omit this ?

> Polarity changes can actually be more frequent than what you describe. Think about a decoder that sits behind an auto-reversing section.

Yes, you are right. Especially in a loco decoder this can obviously happen. And 1 second to recognize the polarity change is definitely to long in this case. Maybe we can reduce the time to 100ms.

> Also, how does this code react in case of zero-stretched zero bits, when using an analog locomotive?

That should be no problem, Bit length until about 60ms are recognized as zero bits. ( unsigned int with micros() ).

tanner87661, 2019-02-18:
You’re right, I mistook it for a line in the ISR.

 

So I gave it a try this morning and it works as expected. It properly decodes standard accessory commands as well as signal commands. Good work.

Hans

kiwi64ajs, 2019-02-21:
I've merged Franz-Peter PR into Master, so if we can try it out on our different hardware and report success, then I'll Tag it as release 2.0.1 so it becomes available to the Arduino Library Manager

MicroBahner, 2019-02-23:
Hi Alex,
my tests with AVR and STM32F1 have been successful.
Franz-Peter

MicroBahner, 2019-05-01:
Hi Alex,
before releasing the new version we should change the following line in nmradcc.h:
`#define NMRADCC_VERSION     200     // Version 2.0.0
`
to 
`#define NMRADCC_VERSION     201     // Version 2.0.1
`
Franz-Peter

kiwi64ajs, 2019-05-01:
Hmmm… ok

Its looks like I have some merge conflicts now…

Will get one of my devs to help me sort it…

Alex

kiwi64ajs, 2019-05-02:
I believe I've sorted the merge conflicts and merged the ESP32 changes into master.

Before I tag the repo to 2.0.1 it would be good if you guys could double check what I've done in case I messed up a merge.

Just noticed one potential difference.
At line 1510 here: https://github.com/mrrwa/NmraDcc/blob/master/NmraDcc.cpp#L1510 we have:

    #ifdef ESP32
        portENTER_CRITICAL(&mux);
    #else
        noInterrupts();
    #endif
        Msg = DccRx.PacketCopy ;
        DccRx.DataReady = 0 ;

    #ifdef ESP32
        portEXIT_CRITICAL(&mux);
    #else
        interrupts();
    #endif

whereas at line: 567 here: https://github.com/mrrwa/NmraDcc/blob/master/NmraDcc.cpp#L567 we just have the ESP32 bit with no normal Arduino noInterrupts() / interrupts() calls
    #ifdef ESP32
    	  portENTER_CRITICAL_ISR(&mux);
    #endif
          DccRx.PacketCopy = DccRx.PacketBuf ;
          DccRx.DataReady = 1 ;
    #ifdef ESP32
    	  portEXIT_CRITICAL_ISR(&mux);
    #endif

 Should these be added?

MicroBahner, 2019-05-02:
Hi Alex,
in line 567 we are within the Dcc-Interrupt, and this interrupt is protected by a flag against interrupting itself. Normally this should never happen, but as you know, we had problems with glitches and that's the reason why this flag was implemented, From normal Arduino point of view there is no need to do a nointerrupt()/interrupt() call at this point.
I don't know the ESP32 and don't even have one. May be it isn't necessary for the ESP32 too.

Franz-Peter

kiwi64ajs, 2019-05-02:
Hi Franz-Peter

> On 2/05/2019, at 9:22 PM, Franz-Peter <notifications@github.com> wrote:
> 
> Hi Alex,
> in line 567 we are within the Dcc-Interrupt, and this interrupt is protected by a flag against interrupting itself. Normally this should never happen, but as you know, we had problems with glitches and that's the reason why this flag was implemented, From normal Arduino point of view there is no need to do a nointerrupt()/interrupt() call at this point.
> I don't know the ESP32 and don't even have one. May be it isn't necessary for the ESP32 too.
> 
Ah ok - that makes sense.

The issue with the ESP32 is the dual core potential for concurrent access that needs to protected against.

So I think we’re ok.

If you can verify its still working ok then I can tag the repo and release the updated version to the Arduino Library Manager

Alex

MicroBahner, 2019-05-02:
Hi Alex,
ok, I'll do same tests the next days.
Franz-Peter

kiwi64ajs, 2019-05-02:
> Hi All,
> Do you guys realize there is more than one implementation of an
> ESP32?
> Expressif (the creators) defined a "ESP32 dev" module that many use,
> but there are several other versions (some better in my opinion like
> the Ezsbc.com board).
> The point is that they do not have a "standard" pinout. In this case
> it does not mean
> physical pin layout, but rather which pins are actually brought out
> to edge points.
>
> So which are you allowing for? Or do you account for all variations?
> Have fun!  :-)
> Best regards,
> Geoff Bunza
>
>
> On 5/2/2019 2:41 AM, Alex Shepherd

Hi Geoff

This relates to enabling compiler support for ESP32 generally and not just for a specific ESP32 board. It’s ESP32 vs AVR vs. STM32

Alex

gbglacier, 2019-05-03:
Understood Alex. I thought this might be worth
      noting. 
      I jst published an ESP32 design (throttle not decoder) and some
      modelers
      did not heed my warning about differences in ESP32's.
      Have fun!  :-)
      Best regards,
      Geoff
      
    
    On 5/2/2019 1:09 PM, Alex Shepherd
      wrote:
    
    Hi
      Geoff
      
      This relates to enabling compiler support for ESP32 generally and
      not just for a specific ESP32 board. It’s ESP32 vs AVR vs. STM32
      
      Alex 
      
      Sent from my iPad

      

      
      "https://github.com/mrrwa/NmraDcc/issues/22#issuecomment-488612175",
      
      "https://github.com/mrrwa/NmraDcc/issues/22#issuecomment-488612175",
      
      thread.
\n\u003e \n\u003e \n\u003e \n\u003e \n\u003e \n\u003e \n\u003e Hi All, \n\u003e Do you guys realize there is more than one implementation of an \n\u003e ESP32? \n\u003e Expressif (the creators) defined a \"ESP32 dev\" module that many use, \n\u003e but there are several other versions (some better in my opinion like \n\u003e the Ezsbc.com board). \n\u003e The point is that they do not have a \"standard\" pinout. In this case \n\u003e it does not mean \n\u003e physical pin layout, but rather which pins are actually brought out \n\u003e to edge points. \n\u003e \n\u003e So which are you allowing for? Or do you account for all variations? \n\u003e Have fun!  :-) \n\u003e Best regards, \n\u003e Geoff Bunza \n\u003e \n\u003e \n\u003e On 5/2/2019 2:41 AM, Alex Shepherd \n\u003e wrote: \n\u003e \n\u003e Hi \n\u003e Franz-Peter \n\u003e \n\u003e \u003e
      [

{

kiwi64ajs, 2019-05-03:
> Understood Alex. I thought this might be worth 
> noting. 
> I jst published an ESP32 design (throttle not decoder) and some 
> modelers 
> did not heed my warning about differences in ESP32's. 

Ah, yes… I guess with the various sources not being too concerned with providing any details or documentation etc its a bit of a murky pond huh…

BTW I’ve finally merged in some ESP32 changes into the NmraDcc library. It compiles and I’m pretty sure I got it right but had to manually resolve conflicts, so if you get a change, it would be helpful if you could check version 2.0.1 in master of the library on some of your AVR and ESP32 project.

Alex

kiwi64ajs, 2019-10-28:
I think this is done

closed 2019-10-28 by kiwi64ajs