mrrwa/NmraDcc #31: Document CV29 output/board addressing mode

chrisridd, 2019-12-14:
Your extension to CV29 could be better documented!

As far as I can see it just changes two things:

1. The way the accessory address is computed
2. Which accessory callback is used i.e. notifyDccAccTurnoutBoard or notifyDccAccTurnoutOutput

Neither of which is very clear - which mode should an accessory implementor use and why?

MicroBahner, 2019-12-14:
Well. I don't see any 'extension' to CV29. The meaning of the bits in CV29 is well documented in the Nmra papers. Bit 6 defines whether the decoder uses board adressing or output adressing. And the corresponding callback is used when an accessory packet is received.

gbglacier, 2019-12-21:
Hi,
    This is just downloaded from the NMRA.org site. Bit 6 is not
      defined. Any change from the current default
      implementation of any addressing scheme should be avoided.
    Configuration Variable 29 Configurations Supported
      Bit 0 = Locomotive Direction: "0" = normal, "1" = rev 215 ersed.
      This bit controls the locomotive's forward and
      backward direction in digital mode only. Directional sensitive
      functions, such as headlights (FL and
      FR), will also be reversed so that they line up with the
      locomotive’s new forward direction. See S-
      9.1.1 for more information.
      Bit 1 = FL location: "0" = bit 4 in Speed and Direction
      instructions control FL, "1" = bit 4 in function group
      220 one instruction controls FL. See S-9.2.1 for more information.
      Bit 2 = Power Source Conversion: "0" = NMRA Digital Only, "1" =
      Power Source Conversion Enabled, See
      CV#12 for more information,
      Bit 3 = Bi-Directional Communications: "0" = Bi-Directional
      Communications disabled, "1" = Bi-Directional
      Communications enabled. See S-9.3.2 for more information.
      225 Bit 4 = Speed Table: "0" = speed table set by configuration
      variables #2,#5, and #6, "1" = Speed Table set by
      configuration variables #66-#95
      Bit 5 = "0" = one byte addressing, "1" = two byte addressing (also
      known as extended addressing), See S 9.2.1
      for more information.
      Bit 6 = Reserved for future use.
      230 Bit 7 = Accessory Decoder: "0" = Multifunction Decoder, "1" =
      Accessory Decoder (see CV #541 for a
      description of assignments for bits 0-6)
      *Note If the decoder does not support a feature contained in this
      table, it shall not allow the corresponding bit to be
      set improperly (i.e. the bit should always contain its default
      value).
    Have fun!  :-)
      Best regards,
      Geoff Bunza
    
    
    On 12/14/2019 5:29 AM, Franz-Peter
      wrote:
    
    
      Well. I don't see any 'extension' to CV29. The meaning of the
        bits in CV29 is well documented in the Nmra papers. Bit 6
        defines whether the decoder uses board adressing or output
        adressing. And the corresponding callback is used when an
        accessory packet is received.
      —
        thread.
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/mrrwa/NmraDcc/issues/31?email_source=notifications\u0026email_token=ADZAHURLTXMIWX4LQ2SMLOLQYTNVJA5CNFSM4J2Z236KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEG4CUFA#issuecomment-565717524",
"url": "https://github.com/mrrwa/NmraDcc/issues/31?email_source=notifications\u0026email_token=ADZAHURLTXMIWX4LQ2SMLOLQYTNVJA5CNFSM4J2Z236KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEG4CUFA#issuecomment-565717524",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]
    
    
      
      
      
      
      
      
      
      
      
      

      
        Geoff Bunza
        gbglacier@comcast.net
        scalemodelanimation.com

MicroBahner, 2019-12-21:
Hi Geoff,
the definitions you posted are the definitions for a multifunction (=locomotiv) decoder. This issue is about an accessory decoder. The meaning of the bits in CV29 ist different between these types of decoders. The meaning of the bits for an accessory decoder is defined later in that paper at pages 10/11:

> Configuration Variable 29 [541] Accessory Decoder Configurations Supported
Bits 0-2 = Reserved for future use.
Bit 3 = Bi-Directional Communications: "0" = Bi-Directional Communications disabled, "1" = Bi-Directional
425 Communications enabled. See S-9.3.2 for more information.`
Bit 4 = Reserved for future use.S-9.2.2
Copyright 1999-2012 by National Model Railroad Association, Inc. Page 11 of 12
Bit 5 = Decoder Type: ‘0’ = Basic Accessory Decoder; ‘1’ = Extended Accessory Decoder
Bit 6 = Addressing Method: ‘0’= Decoder Address method; ‘1’ = Output Address method
Bit 7 = Accessory Decoder: = "0" Multifunction Decoder (See CV-29 for description of bit Assignments for bits
430 0-6), "1" = Accessory Decoder. If bit 7 = 1, then the decoder may ignore the two most-significant
bits of the CV number in Service Mode only. Using this feature CV513 becomes CV1, etc.
Decoders which perform the translation must clearly

There you can see that Bit 6 defines the adressing method.
Regards
Franz-Peter

gbglacier, 2019-12-21:
Hi Franz-Peter,
    Got it. Thanks for contributing to my continuing
        education.
    Whatever way the JMRI default of continuous
        addressing, rather than decoder and "page" or whatever it was
        called, should be left as the default for the NmraDcc library,
        in my opinion. This is as the library was, as well.
    Have fun!  :-)
        Best regards,
        Geoff
    
    
    On 12/21/2019 12:05 PM, Franz-Peter
      wrote:
    
    
      Hi Geoff,
        the definitions you posted are the definitions for a
        multifunction (=locomotiv) decoder. This issue is about an
        accessory decoder. The meaning of the bits in CV29 ist different
        between these types of decoders. The meaning of the bits for an
        accessory decoder is defined later in that paper at pages 10/11:
      
        Configuration Variable 29 [541] Accessory Decoder
          Configurations Supported
          Bits 0-2 = Reserved for future use.
          Bit 3 = Bi-Directional Communications: "0" = Bi-Directional
          Communications disabled, "1" = Bi-Directional
          425 Communications enabled. See S-9.3.2 for more information.`
          Bit 4 = Reserved for future use.S-9.2.2
          Copyright 1999-2012 by National Model Railroad Association,
          Inc. Page 11 of 12
          Bit 5 = Decoder Type: ‘0’ = Basic Accessory Decoder; ‘1’ =
          Extended Accessory Decoder
          Bit 6 = Addressing Method: ‘0’= Decoder Address method; ‘1’ =
          Output Address method
          Bit 7 = Accessory Decoder: = "0" Multifunction Decoder (See
          CV-29 for description of bit Assignments for bits
          430 0-6), "1" = Accessory Decoder. If bit 7 = 1, then the
          decoder may ignore the two most-significant
          bits of the CV number in Service Mode only. Using this feature
          CV513 becomes CV1, etc.
          Decoders which perform the translation must clearly
      
      There you can see that Bit 6 defines the adressing method.
        Regards
        Franz-Peter
      —
      [

{

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/mrrwa/NmraDcc/issues/31?email_source=notifications\u0026email_token=ADZAHURURLXMSTLMBTMLVZDQZZZH7A5CNFSM4J2Z236KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHPCTQQ#issuecomment-568207810",

"url": "https://github.com/mrrwa/NmraDcc/issues/31?email_source=notifications\u0026email_token=ADZAHURURLXMSTLMBTMLVZDQZZZH7A5CNFSM4J2Z236KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHPCTQQ#issuecomment-568207810",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]
    
    
      
      
      
      
      
      
      
      
      
      

      
        Geoff Bunza
        gbglacier@comcast.net
        scalemodelanimation.com

MicroBahner, 2019-12-21:
Hi Geoff,
yes, I think we're not going to change this behavior of the library. When initializing the library you can set the default adressing mode as you want. If you don't want it to be changed later, you have to inhibit changing the bit in CV29. That must be done in the sketch.

As I understood, the question in this issue was about a better documentation. But the library complies with the standard - also in this respect. I have therefore pointed out that it is not an extension. In my opinion we cannot describe the standards in the library documentation.

Best regards
Franz-Peter

chrisridd, 2019-12-22:
Like Geoff, I'd only found the definition for CV 29 for locomotive decoders - but now I can see the different CV 29 definitions for accessory decoders I concur that this library's use of bit 6 is quite standard and correct and does not need changing.

However I still do not understand what the mode actually means, which was my main question. Do you have a reference to the NMRA docs that explains it?

MicroBahner, 2019-12-22:
Hi,
The difference between output addressing and decoder addressing is described in the same NMRA document (9.2.2). This can be found in the description of the CV's 513 and 521 which define the decoder address (on pages 9/10). But in my opinion the description is a little bit hard to understand..
A graphic, which should make things clearer, can be found [here](https://www.opendcc.de/info/decoder/cv513_cv521.gif) (its on a german website).

Decoder addressing is rather historical. It is based on the first decoders, which could always control 4 turnouts. The decoder was addressed via the 'main address', and the turnouts via a sub-address(0..3). Such decoders still exist.
Output adressing combines both addresses and the turnouts/signals are numbered continuously. This is more flexible. A decoder can control an arbitrary number of accessories. The adress in CV513/521(1/9) is usually the first in the line.

kiwi64ajs, 2020-01-10:
I don't think there is an issue here apart from the DCC Standards being not precise enough, but I believe the code is doing the right things in terms of handling the Output Addressing Mode

closed 2020-01-10 by kiwi64ajs