mrrwa/NmraDcc #33: GPL vs MIT license

vsky279, 2020-01-31:
Dear all,

I very much appreciate your work. This also why I took your driver and ported it in the [NodeMCU Lua](https://github.com/nodemcu/nodemcu-firmware) a Lua based interactive firmware for ESP8266, ESP8285 and ESP32. It has been [merged](https://github.com/nodemcu/nodemcu-firmware/pull/2905) already in the `dev` branch of the project. It works very well and I'm glad I can play with NMRA DCC in Lua.

However one of the senior developpers has recently raised an issue ([#3025](https://github.com/nodemcu/nodemcu-firmware/issues/3025)) pointing out that your code is distributed under GPL license. The NodeMCU Lua project is distributed under MIT license which I understand is a bit wider. Including GPL code would require the project to be under GPL which not something that is on the table.

Therefore I would like to ask you to consider relicensing your great project under **MIT license** :pray:. It would allow spreading it further into the world :smile:.

Thank you. Best regads,

bobjacobsen, 2020-01-31:
The [license is actually the LGPL 2](https://github.com/mrrwa/NmraDcc/blob/master/LICENSE), not the pure GPL. For discussion of the difference, see [here](https://en.wikipedia.org/wiki/GNU_Lesser_General_Public_License#Differences_from_the_GPL) and [here](https://opensource.stackexchange.com/questions/6405/lgpl-licensed-code-reuse-in-mit-licensed-project).

If you want to use the LGPL'd NmraDcc with a MIT licensed project, i.e. NodeMCU Lua, you have to use NmraDcc as a library.  Make the necessary changes either here in the mrrwa/NmraDcc or an appropriately licensed fork, then create binary libraries and include files that can be distributed to and referenced by the other project(s).

kiwi64ajs, 2020-02-03:
Hi Guys,

I've just corrected the header text in the NmraDcc.[hc] files to be the "Lesser GPL" which is consistent with the original intent and committed the changes to the master branch in the Git repo.

I've also responded to this issue here: https://github.com/nodemcu/nodemcu-firmware/issues/3025#issuecomment-581324137

HTH
Alex Shepherd

closed 2020-02-03 by kiwi64ajs