mrrwa/NmraDcc #34: Hello is there a way to use and enter more than 28 functions?

ajoreis, 2020-02-20:
Hello is there a way to use and enter more than 28 functions?

I would like to use more than the currently available functions 28 for a sound module, according to the RCN-212 DCC protocol
Operating commands for vehicle decoders Edition 01.12.2019 R
http://normen.railcommunity.de/RCN-212.pdf
Can be used max F68.
2.3.4 Function control F13-F68
These commands are two bytes long and have the following formats:
 Function control F13-F20: 1101-1110 DDDD-DDDD
 Function control F21-F28: 1101-1111 DDDD-DDDD
 Function control F29-F36: 1101-1000 DDDD-DDDD
 Function control F37-F44: 1101-1001 DDDD-DDDD
 Function control F45-F52: 1101-1010 DDDD-DDDD
 Function control F53-F60: 1101-1011 DDDD-DDDD
 Function control F61-F68: 1101-1100 DDDD-DDDD

Is there any way to extend it?

Thank you for an info

Olaf Reis

kiwi64ajs, 2020-02-20:
Hi Olaf,

> On 21/02/2020, at 8:24 AM, ajoreis <notifications@github.com> wrote:
> 
> Hello is there a way to use and enter more than 28 functions?
> 
Where there’s a will there’s usually a way...
> I would like to use more than the currently available functions 28 for a sound module, according to the RCN-212 DCC protocol
> Operating commands for vehicle decoders Edition 01.12.2019 R [...]
I just had a quick look and I expect we “just” need to add some more function group decoding logic.

I don’t have anything that can generate those extended packets but may be able to do some code changes if you can do the testing.

Alternatively would you be able to make the changes and contribute those back to the project?

Regards

Alex Shepherd

devel-bobm, 2020-02-20:
Olaf,

It would appear that the protocol you reference is making use of "Future Expansion" instructions in the NMRA specification 9.2.1.  In the electronic copy of that NMRA spec 9.2.1 that I have, the particular reference would be line 94.  I do not know if I have the latest version; the one I have shows a date of July 2012 on the first page.  Any newer issue of that spec might have different line numbering, or even different instruction definitions.  

Other existing NMRA specs, or even future NMRA specs, could sneak in a few new instructions without requiring modification to 9.2.1, so following the RCN-212 protocol is somewhat risky.

If you just want to do something for personal use, and you can live with the idea that _your_ choice of mapping might conflict with some future NMRA-compatible instruction(s), you could make something work.

Does the MRRWA NMRA Dcc library decode the instructions as specified in RCN-212?  No.  Could it?  Sure.  You could rework the library to make it do so.  

If you do make those changes, should you submit your changes for inclusion in the MRRWA library?  Sure.  Would your changes be included in the library distribution files?  That is a question that I cannot answer.  Such a question would need some careful thought by those who are running the show.

Regards,
Billybob

http://normen.railcommunity.de/RCN-212.pdf

kiwi64ajs, 2020-02-20:
> On 21/02/2020, at 9:58 AM, Bob Milhaupt <notifications@github.com> wrote:
> 
> Olaf,
> 
> It would appear that the protocol you reference is making use of "Future Expansion" instructions in the NMRA specification 9.2.1. In the electronic copy of that NMRA spec 9.2.1 that I have, the particular reference would be line 94. I do not know if I have the latest version; the one I have shows a date of July 2012 on the first page. Any newer issue of that spec might have different line numbering, or even different instruction definitions. 
> 
> Other existing NMRA specs, or even future NMRA specs, could sneak in a few new instructions without requiring modification to 9.2.1, so following the RCN-212 protocol is somewhat risky.

Yeah, I think I’d wrap the changes in conditional defines to allow us to toggle it on/off in the future if the NMRA and RCN disagree on future expansion specs… ;)

#define RCN_212
#ifdef RCN_212
#endif

Regards

Alex Shepherd

ajoreis, 2020-02-21:
Hello everybody,

I would like to thank you for your quick answers.
Unfortunately I am not a good programmer to carry out the expansion of Liberia, it is definitely not easy to program this extension.
I have another Ruud Boer DCC library.
https://github.com/beryou/dcc-master/tree/master/Arduino_DCC_S88/libraries
This should support the extended commands but is not as extensive as your libery. I take these for testing and if I have success then I would contact you again.

You shouldn't invest any additional work there.

I think the idea with the Variabe regarding the extension is very good.
#define RCN_212
#ifdef RCN_212
#endif

PS. my english i don't very good sorry

Thank you very much

Olaf

kiwi64ajs, 2021-04-04:
In the absence of someone with the appropriate hardware to code and test this extension I'm closing this

closed 2021-04-04 by kiwi64ajs