mrrwa/NmraDcc #36: DCCInterface_TurntableControl question

Chemdawgds, 2020-03-16:
I have a very big interest in this example.  My hardware is a bit different though.  I have my dcc interface going to my arduino uno, along with a TB6600 stepper motor driver.  When I load the sketch for this, and open the serial monitor.  I can see that the arduino is getting the dcc responses.  However mo stepper is not moving.  I can feel that it's on.  It just doesnt move.  Ideas?

MicroBahner, 2020-03-16:
Can you show a schematic of your hardware? That would help answer your question.

Chemdawgds, 2020-03-16:
I apologize.  I thought I had attached a picture of the dcc board.  I dont 
have a schematic of the Tb6600.  However.  There are your two power, the 4 
wires to the stepper, then you have pulse plus and minus, direction plus 
and minus, and enable plus and minus inputs.  The arduino is the uno.

I had the plus for the pulse, direction, and enable going to 4,5,6 input 
pins on arduino.  Pin 3 is the home switch.  Pin 2 is the dcc signal from 
the dcc board.  All the minus wires from the driver go to the arduino minus 
pin.

I hope this will help.

kiwi64ajs, 2020-03-17:
Hi Patrick,

Try commenting out line 157 of the DCCInterface_TurntableControl example sketch:

	https://github.com/mrrwa/NmraDcc/blob/master/examples/DCCInterface_TurntableControl/DCCInterface_TurntableControl.ino#L157

which is:

	stepper1.setPinsInverted(false, false, true); // Its important that these commands are in this order

as this changes the default stepper pin polarity to what was needed for the A4988 but the TB6600 is probably different.

What this will do is leave the pin polarity within the library as default which is not-inverted, which should hopefully work with the TB6600

HTH

Alex Shepherd

Chemdawgds, 2020-03-17:
So I was on my bench last night and I tried something.  I am loved pins 4 5 
6 to 8 9 and 10, and it worked.  So looking at your email here about 
commenting out the particular code, I think you are right.  I haven't got 
to try it yet but I will.  I will let you know.  Thank you.

Chemdawgds, 2020-03-17:
Hey Alex.  So I commented out that line, and there was no difference.  So I 
moved pins 4,5,6 to 8,9,10 and it works great.  Now dont quote me on this, 
but I believe the first 8 pins are pwm?  The TB6600 uses that H bridge 
setup.  So I believe it's a voltage thing.  Everything works very well on 
8,9,10.  Only thing I would like to do is figure out how to lock the rotor. 
 Then the stepper gets to the spot it is going and stops, you can turn it 
freely.  I'm not sure how you would set it up so when it gets there its locked.

I want you to know how much I appreciate the help.  I got zero help from 
the arduino forum, and a few other places.  Really I was asking for a push 
in the right direction.  Thank you.

Pat

closed 2020-10-19 by kiwi64ajs