mrrwa/NmraDcc #37: Problem with MultiFunctionDecoder

christianloesel, 2020-03-17:
Hello,

i have the problem that in my project the notifyDccSpeed callback never gets raised.
Further the decoder doesn#t set its address correctly. I did help out with Flag FLAGS_AUTO_FACTORY_DEFAULT. But if I don't set this flag, any random value will be returned from CV 1.
Can anyone help me?

My project is hosted here: https://github.com/christianloesel/Messwagen-NMRADCC/tree/master/Neigungsmesswagen

Thanks

closed 2020-03-26 by christianloesel

kiwi64ajs, 2020-03-29:
Hi Christian,

Hmmm... Looking at your code its not obvious what could be wrong.

Do you receive any DCC packets and/or does any of the notifyXXX call-back functions get called?

Can you email me a PDF of the schematic so I can see more about what you’re trying to do?

Regards

Alex Shepherd

christianloesel, 2020-03-30:
I solved this. It was simply an error in the hardware :-(
That's why i closed this issue.

Thank you for this great library!!!