mrrwa/NmraDcc #42: More than one locomotive / function decoder address

ajoreis, 2020-06-17:
Hallo,,
is there a possibility that you can use more than one locomotive address. e.g. Address 100 is configured, the second address is then address 101 and possibly a third one with address 102 etc.

Thanks for your Info.

Olaf

MicroBahner, 2020-06-17:
Hi Olaf,
if you don't use the flag 'FLAGS_MY_ADDRESS_ONLY' you get a callback at every received address. You can than select the wanted addresses by yourself.
Regards
Franz-Peter

ajoreis, 2020-06-17:
Hi Franz-Peter,
Ok if I see that correctly this is the init string:
Dcc.init (MAN_ID_DIY, 100, FLAGS_MY_ADDRESS_ONLY, 1);
then what can I use as a variable? or can add a second by adding this variable with 1 e.g.
FLAGS_MY_ADDRESS_ONLY2 = (FLAGS_MY_ADDRESS_ONLY + 1)

I use this function:
void notifyDccFunc (uint16_t Addr, DCC_ADDR_TYPE AddrType, FN_GROUP FuncGrp, uint8_t FuncState)
Which variable is the decoder address here?

christianloesel, 2020-06-17:
The FLAGS_MY_ADDRESS_ONLY ist used to raise the callbacks only when your given address is called. 

If you leave it, they will always be called and you can filter the „Addr“ parameter in the callbacks. 

Greetings Chris

MicroBahner, 2020-06-17:
Hi,
the decoder address is coded in CV values, not in the init flags. The flag FLAGS_MY_ADDRESS_ONLY means the lib compares the received address to the addres in the CV and the notifyDccFunc is only called if this matches. If you don't set this flag, this comparison is not done and the notifyDccFunc is called with every address. You have to compare the addresses in your sketch.
Thi init command must look like this:
Dcc.init( MAN_ID_DIY, 100, 0, 0 );

It's yust as it Chris wrote ;)

kiwi64ajs, 2020-06-17:
Hi Olaf,

> On 17/06/2020, at 8:29 PM, ajoreis <notifications@github.com> wrote:
> Hi Franz-Peter,
> Ok if I see that correctly this is the init string:
> Dcc.init (MAN_ID_DIY, 100, FLAGS_MY_ADDRESS_ONLY, 1);
> then what can I use as a variable? or can add a second by adding this variable with 1 e.g.
> FLAGS_MY_ADDRESS_ONLY2 = (FLAGS_MY_ADDRESS_ONLY + 1)
> 
No, the FLAGS_MY_ADDRESS_ONLY flag enables a filter that compares the address in all DCC Packets to the DCC address stored in the EEPROM Address CVs as per CV29. If the address matches you call-back will be called.

If you omit the FLAGS_MY_ADDRESS_ONLY flag like:

	Dcc.init (MAN_ID_DIY, 100, 0, 0);

Then the call-back functions will get called for any packet of the appropriate type for all/any addresses. There will not be any address filtering.
> I use this function:
> void notifyDccFunc (uint16_t Addr, DCC_ADDR_TYPE AddrType, FN_GROUP FuncGrp, uint8_t FuncState)
> Which variable is the decoder address here?
> 
The address will be in the Addr (1..9999) parameter and the type of address will be in the AddrType (DCC_ADDR_SHORT or DCC_ADDR_LONG) parameter.

HTH

Alex Shepherd

ajoreis, 2020-06-19:
Hello Alex
Thank you for your info, I've tried a few things, but unfortunately I can't get it. I just send you my little sketch, in this sketch I create a small sound decoder for a train station, because I am restricted to 29 with the functions and I have enough space on the memory stick, I would like to play more than only 29 sounds. that's why I considered using several consecutive Loc addresses. Its functions are then assigned to the sound file.
[NmraDccMultiFunctionDecoder_2_MP3.zip](https://github.com/mrrwa/NmraDcc/files/4804916/NmraDccMultiFunctionDecoder_2_MP3.zip)

MicroBahner, 2020-06-19:
Hi Olaf,
you still use the 'FLAGS_MY_ADDRESS_ONLY' Flag when initializing the Lib. You must not do that, if you want to use more than one address. You must initialize with
`Dcc.init( MAN_ID_DIY, 100, 0, 1);`
You have to check the addresses by yourself in the notifyDccFunc() function.

regards
Franz-Peter

kiwi64ajs, 2020-10-19:
Has this issue/request been resolved?

closed 2021-04-04 by kiwi64ajs