mrrwa/NmraDcc #45: [Question] About interrupts on an ATMega328

mijnmodelbaan, 2020-11-22:
Hi,

Great library you all created. Thanks!

My question is, can I use other pins than the 'standard' one (D2 on the Arduino) for inputting a DCC signal. F.i. the ATMega328P has a max of 23 programmable I/O pins all with the possibility of making them generate an interrupt. Say I want to use PD6 as input; this pin generates PCINT22 which would generate a jump to the PCI2 vector (and back). Is this already implemented in the library?

Thanks for reading/answering this,
Willem.

kiwi64ajs, 2020-11-22:
Hi Willem,

Currently the library only supports the the “core Interrupts” (not sure the correct term) and not “Pin Change” interrupts. I know there are some wrapper libraries that make the use of these interrupts easier or transparent but it also introduces additional latency as the logic has to check each PCInt source to see which pin changed, to know which call-back to call.

You’re welcome to give it a try and submit a patch but you’d need to revalidate the DCC bit-timings to make sure you’ve not broken the timing as we’ve worked hard to get the core library validated by the NMRA Conformance Testing Team and its now working well and a commercial decoder based on this codebase have passed the NMRA Tests.

Hence I’d be a little hesitant to mess with this part of the code right now...

HTH

Alex Shepherd

mijnmodelbaan, 2020-11-26:
Hi Alex,

Thanks for the reply. I've been working on a bare version of the library (so only for ATMega328) and got it working to my needs, so with the PinChangeInterrupt implemented.

Now I'm working on another version with the complete library as it is now. See if I can get that working, too. Will be a little different from the above version as I don't want to introduce a lot of extra stuff, keep it as close to the original as possible.

Willem.

mijnmodelbaan, 2020-11-30:
Hi Alex,

Finished the version mentioned in my previous post, the one with the complete library.
You can find the .h file [here](https://github.com/mijnmodelbaan/justforsaving/blob/master/NmraDcc.h) and the .cpp file [here](https://github.com/mijnmodelbaan/justforsaving/blob/master/NmraDcc.cpp).
I want/need to do more testing. I only used one of your examples for testing and I've no idea what the effect is on the timing etc.

The files should work as they are, one only has to put the correct pin number in the calling program; #define DCC_PIN x
It should work with all the available I/O pins on an ATMega328P, digital and analog.

I've ordered some components and when they arrive I'll do some more testing and try to find out about the timing.

Willem.

kiwi64ajs, 2021-04-04:
Hi Willem,
Have you made any more progress with testing your proposed changes? If yes then reopen this issue

closed 2021-04-04 by kiwi64ajs