mrrwa/NmraDcc #48: ESP32 version stops working when loosing interupts or signal is bad

Jueff, 2021-03-10:
sometimes ESP32 runs into the problem, that no DCC data are processed anymore.

I found out that in error state we have

- DccRx.State = WAIT_PREAMBLE
- ISRLevel = 0
- ISRWatch = 1

so this code 
```
#ifdef ESP32
	ISRWatch = ISREdge;
#else
attachInterrupt( DccProcState.ExtIntNum, ExternalInterruptHandler, ISREdge );
// enable level checking ( with direct port reading @ AVR )
ISRChkMask = DccProcState.ExtIntMask;       
ISRLevel = (ISREdge==RISING)? DccProcState.ExtIntMask : 0 ;
#endif

```

because ISRChkMask and ISRLevel  are not set correctly (in ESP32 case)

The fixed code could look like this - but maybe it's not the only location where levels aren't set correctly.
```
#ifdef ESP32
    ISRWatch = ISREdge;
#else
    attachInterrupt( DccProcState.ExtIntNum, ExternalInterruptHandler, ISREdge );
#endif
// enable level checking ( with direct port reading @ AVR )
ISRChkMask = DccProcState.ExtIntMask;       
ISRLevel = (ISREdge==RISING)? DccProcState.ExtIntMask : 0 ;

```

MicroBahner, 2021-03-17:
Hi Jueff,
You are right. The two lines of code after 
`// enable level checking`
should be active for all processors. So the `#endif` must be shifted 3 lines up.

In all other cases where the ISR edge is changed and this two variables are set it is correct.

Regards
Franz-Peter

kiwi64ajs, 2021-04-04:
I've committed this change - thanks

closed 2021-04-04 by kiwi64ajs