mrrwa/NmraDcc #51: Support for Emulated Storage

billswartz7, 2021-03-29:
I have a version where I ported the code to Seeed Studio XIAO. The XIAO only has emulated storage.   I have implemented a conditional compile so the EEPROM code works for getting and setting CVs. Attached are files. I also made changes to FlashStorage.h so that the read operations would be very efficient.    Other MCUs can be added using a similar technique.

[FlashStorage.h.txt](https://github.com/mrrwa/NmraDcc/files/6222983/FlashStorage.h.txt)
[NmraDcc.h.txt](https://github.com/mrrwa/NmraDcc/files/6222985/NmraDcc.h.txt)
[NmraDcc.cpp.txt](https://github.com/mrrwa/NmraDcc/files/6222986/NmraDcc.cpp.txt)

kiwi64ajs, 2021-04-04:
Looking at the file: FlashStorage.h.txt it looks to have been an early version of this project: https://github.com/cmaglie/FlashStorage so it makes me wonder why don't we modify the code to use EEPROM if it is available or use the latest version of this library and get the benefits of the later supported chips without any EEPROM

kiwi64ajs, 2021-04-04:
Also you seem to have also included a function: bool DCC_present( int adc_full_scale ) that I'm not sure what its for

billswartz7, 2021-04-04:
I totally agree.   This was my intention.   

I added the DCC_present as a convenience function.    I wanted a cheap and dirty solution to see if the decoder was connected to DC or DCC.   It could have been written using a timer but I wrote it so that if something hung I might avoid be able to detect things quickly especially if I am working on a new device.    There is no need to include it.

kiwi64ajs, 2023-02-11:
I've added support for the EEPROM Emulation library: FlashStorage_SAMD in version 2.0.14 of the NmraDcc library, which should resolve this issue in the short term. It would be good to refactor persistent storage to a more flexible solution but its not my top priority right now

closed 2023-02-11 by kiwi64ajs