mrrwa/NmraDcc #59: NMRADCC on Attiny85 stops working over time

PrecooledMunk, 2021-08-29:
Hi. i'm using the NmraDCC lib. on an Attiny85 (tried several different Attiny85 boards), to control 2 relays as a reverse loop module. I'm using the turnout function of the NmraDCC lib. and it works flawlessly for like an hour, where after the module stops responding to the DCC commands (it may stop all commands / codehandling im not sure). A powercycle of the Attiny85, solves the problem, again for some time. As a quick powercycle helps, I don't think it's a heat issue. As mentioned I have tried several types om presoldered/preassembled Attiny85 boards and the result is the same. Has any of You experienced anything similar ? Is there a way of maybe resetting the Attiny85 codewise like every 30 min, or maybe every time  the relays have been switched off ? (i know this is not elegant).
I think the Attiny85 has a perfect formfactor and handling cap. for a reverse loop module, so would really like to stay on this small platform.

kiwi64ajs, 2021-08-30:
Hi,

My guess is the relays switching on/off are interfering with the normal operation of the ATTiny85.

Can you give any more details about how things are wired and more importantly how you’re handling the Back EMF from the relays? Perhaps a wiring diagram (sketch on paper) and photos?

One thing you can do is to enable the WatchDog timer that logic in the loop has to keep resetting and if its fails to do that, thw WatchDog timer will fire and reset the ATTiny85.

Here is the bare-bones example of what you need to add:

#include <avr/wdt.h>

void setup() {

  // Enable WatchDog for 30ms timeout
  wdt_enable(WDTO_30MS);
}

void loop() {

  // Reset the WatchDog every loop
  wdt_reset();
}

HTH

Alex Shepherd

PrecooledMunk, 2021-08-30:
Hi Alex, and thanks for Your quick reply.

The EMC could defenitly be an issue. The relaymodule and the Attiny85's, are cheap chinese circuits, and I have not done anything in perticular to minimize EMC. The outputs of the Attiny85 are wired directly to the relaymodule. The relay module reacts to the outputsignal going to ground, so as not to fry the outputs of the Attiny85.

I won't have access to the circuit untill this thursday evening, where I will try Your WatchDog code snippet, aswell as take some pictures and do a wiring diagram.

Br. Anders Persson

closed 2023-02-07 by kiwi64ajs