mrrwa/NmraDcc #64: Running NmraDcc on ESP8266

pelestedt, 2022-10-20:
According to the Arduino IDE, when installing NmraDcc, the package supports ESP8266. Now trying to run on an ESP8266, Lolin Wemos D1R2, we fail to get any DCC messages decoded. The program and shield works fine on an UNO. Can you assist?

kiwi64ajs, 2022-10-20:
Hi pelestedt,

> On 21/10/2022, at 7:50 AM, pelestedt ***@***.***> wrote:

> According to the Arduino IDE, when installing NmraDcc, the package supports ESP8266. Now trying to run on an ESP8266, Lolin Wemos D1R2, we fail to get any DCC messages decoded. The program and shield works fine on an UNO. Can you assist?

Yes, but I need a bit more information to be helpful as works on a UNO but not on an ESP8266 isn’t particularly helpful… ;)

Can you please answer the following questions:

- What is the DCC interface circuit you are using? Is there a circuit or board schematic you can point me to?

- Does it have an external Pull-Up resistor or does it rely on the uController to provide the Pull-Up?

- What pin on the ESP8266 is being used for the DCC Signal input?

- Does this pin support Hardware Interrupts?

- Can you provide the Arduino sketch you are using to test the ESP8266?

Regards

Alex Shepherd

m: +64-21-777764
e: ***@***.***

pelestedt, 2022-10-21:
Hi Alex,

Thanks for your quick response.
I have done the "shield" I am using myself and it has a 10k pullup on the I/O pin.
I have done a board in KiCad. I can send you the drawing if you want.

I have used your example files to test (changing the I/O pins to Dn first)

What I think you suspect and I believe I have now confirmed is that I use the only GPIO on the ESP8266 that does not support interrupt, D0.

Searching on internet gives mixed answers. Some say all GPIO support interrupt other that all except D0 (GPIO16) does.
I will change to D1 and test.
I will let you know the outcome of that.

Thanks again for your support and hinting me in the right (hopefully) direction.

Best regards
Peter Elestedt

kiwi64ajs, 2023-02-07:
@pelestedt Did using a different pin to D0 resolve your issue?

pelestedt, 2023-02-07:
Hi!
Yes that solved it!

closed 2023-02-07 by kiwi64ajs