mrrwa/NmraDcc #66: Compatibility with LGT8F328P

kamilbystryk, 2023-01-19:
Is the library compatible/works correctly with chinese clone of AtMega - LGT8F328P?

devel-bobm, 2023-01-19:
If the board is truely a clone, then it should be compabible.

kiwi64ajs, 2023-01-19:
Hmmm… I didn’t know about these chips or boards.I’ve just had a quick Google for them and from what I read I would expect it to work.Have you been able to try it yet?I’d have to order some to try it. RegardsAlex Shepherd

kamilbystryk, 2023-01-20:
Currently I cant check in practise - i'm waiting for sample and I will try it. 
Chinese processor isn't true clone, have some difference to oryginal Atmega (including another SDK/HAL).

devel-bobm, 2023-01-20:
> Chinese processor isn't true clone, have some difference to oryginal Atmega

So it will depend entirely upon how well the chinese crap happens to mimic the ATMEL design.  I'd suggest that this "issue" belongs in the MRRWA mailing list, not in the Git repository!

kamilbystryk, 2023-02-27:
Now I can confirm - library on LGT8F328P works correctly.

kiwi64ajs, 2023-02-27:
Excellent, good to know.

I ordered a few from eBay, but they haven’t arrived yet so I’ve not been able to try.

Alex

kiwi64ajs, 2023-02-27:
Confirmed that it works on this chip