mrrwa/NmraDcc #67: Hello, I saw this reply and was very happy to see that someone else was using a XIAO. I am trying to get this compiled using the files listed here, but I keep running the compiler telling me that the class FlashClass has no member named 'read_offset'. I have the Nmra files in the Nmra folder in libraries, and the FlashStorage file in the library for the XIAO. I did try loading the FlashStorage library and putting it there with no help. My appologies for the newb question, but we all gotta start somewhere. Thanks!

gargoylien, 2023-01-23:
I have a version where I ported the code to Seeed Studio XIAO.   I have implemented a conditional compile so the EEPROM code works for getting and setting CVs.    Attached are files.   I also made changes to FlashStorage.h so that the read operations would be very efficient.   The XIAO only has emulated storage.    If can be backwards compatible so only one library is needed.

[NmraDcc.h.txt](https://github.com/mrrwa/NmraDcc/files/6222955/NmraDcc.h.txt)
[NmraDcc.cpp.txt](https://github.com/mrrwa/NmraDcc/files/6222956/NmraDcc.cpp.txt)
[FlashStorage.h.txt](https://github.com/mrrwa/NmraDcc/files/6222958/FlashStorage.h.txt)

_Originally posted by @billswartz7 in https://github.com/mrrwa/NmraDcc/issues/38#issuecomment-809499778_

gargoylien, 2023-01-28:
Hi again. I figured out that the library FlashStorage.h was not in the correct spot. I reloaded everything from scratch and put the files in and successfully compiled. I am still unable to get the chip to accept any input from the DCC signal though. I'm trying to do some debugging on it, but not having much luck. Any help would much be appreciated.

gargoylien, 2023-02-07:
> I've just pushed an update to the library version 2.0.14 that will use the
> FlashStorage_SAMD library to provide emulated EEPROM.
> I was able to build and run the NmraDccMultiFunctionMotorDecoder sketch
> and compiled and ran ok and displayed DCC Packets as I change the throttle
> settings for DCC Loco Short Address 3.
> Please update the library when version 2.0.14 is available and try it

Thanks for the update. I’ve been messing around with that library too, and
had some success, albeit with bugs. I’ll try your new code and see how it
goes.
Thank you very much for taking the time to do this.

Sean

gargoylien, 2023-02-10:
Well, still no joy. I can make the decoder work by commenting out the "DECODER_LOADED", but even then only function 0 and speed work. Changing values in any other function don't work at all. If I uncomment the DECODER_LOADED, it does nothing. I've tried a few different of the sketch examples without any luck. I tested the XIAO with a flash write/read sketch and that seems to work fine.

kiwi64ajs, 2023-02-11:
> Well, still no joy. I can make the decoder work by commenting out the "DECODER_LOADED"

Sorry, my testing was using the sketch: NmraDccMultiFunctionDecoder_1 and not the one that you were using. but I've switched the one you were using and tweaked it to work with the Xiao and seemed to work ok.

Your comments about "DECODER_LOADED" don't make sense to me as I can't find what you are referring to. Could this be something you have added and if so can you email it to me.

> but even then only function 0 and speed work. Changing values in any other function don't work at all.  If I uncomment the DECODER_LOADED, it does nothing. I've tried a few different of the sketch examples without any luck. 

The sketch was only ever intended to demonstrate motor speed and direction and directional headlight function. No other functions were supported in the sketch.  

> I tested the XIAO with a flash write/read sketch and that seems to work fine.

Ok, good. That was the main objective of my code changes in the library.

gargoylien, 2023-02-11:
I’m using variants of the NMRA/SMA sketches, and all of those have DECODER_LOADED defined if you want the sketch to retain the CV values upon reload. This is the part that requires the EEPROM emulation to work correctly. I appreciate you working on this as it has definitely made a big difference on how close I am to having a fully functional decoder. I’ll keep tinkering with it here and see what I can find. Meanwhile, could you possibly share the sketch you are using so I can make sure I don’t have other problems?

Thanks
Sean

kiwi64ajs, 2023-02-15:
I have just added the example: NmraDccMultiFunctionMotorDecoder-XIAO-Expansion which will run on the Seeed Xiao + Xiao Expansion board and IoTT DCC Communication Interface connected to the Grove Analog Port and displays Throttle actions on the OLED display

closed 2023-02-15 by kiwi64ajs

gargoylien, 2023-02-15:
Thanks Alex!
I’ll check that out to see what I have wrong.

gargoylien, 2023-02-18:
I used the sketch NmraDccMultifinctionDecoder from NMRA examples, and I was
able to alter it to get everything working. I think this version of the
NMRA code works fine with XIAO now, so thank you for your work on getting
this board added.

Sean