mrrwa/NmraDcc #68: Cant get runing on RP2040 (Pi Pico)

Ecksteher, 2023-02-05:
Hello i trying to get it runing on My Pi Pico. i Wred a fiew times "ARDUINO_ARCH_RP2040" so i thinked it wud work. but as i uploaded my simple sniffer sketch it just output "Sniffer Start". and dthats it.
I tried also on an Arduino Nano and it worked.

```
#include <NmraDcc.h>
#define DCC_PIN   2
#define DCC_DECODER_VERSION_NUM 11  // Set the Decoder Version - Used by JMRI to Identify the decoder
NmraDcc  Dcc ;
DCC_MSG  Packet ;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Dcc.pin(0, DCC_PIN, 1);
  Dcc.init( MAN_ID_DIY, DCC_DECODER_VERSION_NUM,  FLAGS_DCC_ACCESSORY_DECODER , 0 );
  while (!Serial) {

  }
  Serial.println("Sniffer Start");

}

void loop() {
  // put your main code here, to run repeatedly:
  Dcc.process();
}
void notifyDccMsg( DCC_MSG * Msg)
{
  Serial.print("notifyDccMsg: ") ;
  for (uint8_t i = 0; i < Msg->Size; i++)
  {
    Serial.print(Msg->Data[i], HEX);
    Serial.write(' ');
  }
  Serial.println();
}
```

On the Hardware side i used bridge rectifier for Ground and a voltage devider that converts the rail voltage to 3.3V claped wit a zener diode and feed it through an schmittrigger. than into my Controller.

Cann somone help me pleas?

kiwi64ajs, 2023-02-07:
Hi @Ecksteher 
I think the problem is the definition of the DCC Input Pin. Try changing: 

  Dcc.pin(0, DCC_PIN, 1);

to:

  Dcc.pin(DCC_PIN, 1);

Doing the above code change makes your code work for me on a RPI-Pico

Ecksteher, 2023-02-07:
Yes this solved my problem. but i tried bevore and it din't run before, probably I changed somting else to.
Thank you verry much.

closed 2023-02-07 by Ecksteher