mrrwa/NmraDcc #70: platfromio 6.1.6 Add NMRADCC library No corresponding EEPROM library found

murarduino, 2023-03-06:
I developed NMRADCC based on EPS32 and added NMRADCC in Platfromio (version:2.0.15) library, but during compilation, it was discovered that the EEPROM library could not be found.
I found a lot of EEPROM related libraries in the PlatformIO library manager, and I 'm not sure which one to choose.

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/denky_d4.html
PLATFORM: Espressif 32 (4.1.0) > Denky D4 (PICO-V3-02)
HARDWARE: ESP32 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20001.0 (2.0.1)
 - tool-esptoolpy @ 1.30100.210531 (3.1.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- NmraDcc @ 2.0.15
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- FS @ 2.0.0
Building in release mode
Compiling .pio\build\denky_d4\src\main.cpp.o
Compiling .pio\build\denky_d4\lib2a9\NmraDcc\NmraDcc.cpp.o
.pio/libdeps/denky_d4/NmraDcc/NmraDcc.cpp:53:10: fatal error: EEPROM.h: No such file or directory

****************************************************************
* Looking for EEPROM.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:EEPROM.h"
* Web  > https://registry.platformio.org/search?q=header:EEPROM.h
*
****************************************************************

 #include "EEPROM.h"
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\denky_d4\lib2a9\NmraDcc\NmraDcc.cpp.o] Error 1

kiwi64ajs, 2023-03-06:
Hmmm... I've never tried to use Platformio so I'm unlikely to be of much help but does this solution mean anything to you?
https://community.platformio.org/t/missing-eeprom-h/23547

murarduino, 2023-03-06:
> 

Thanks for the tip. I tried to force the EEPROM in lib_ldf_mode and the compilation passed.
Perhaps there was a problem with the PlatformIO update synchronizing with the framework-arduionoespressif32 library, however, forcing the addition of the EEPROM will get the platform compiled.

platfromio.ini:
[env:denky_d4] 
platform = espressif32
board = denky_d4 
framework = arduino 
lib_deps = EEPROM ;forced addition
            mrrwa/NmraDcc@^2.0.15

closed 2023-03-06 by murarduino