mrrwa/NmraDcc #73: EEPROM.commit causes "Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)."

PontusO, 2023-07-10:
As described in the title when running the unmodified library and example on an ESP32C3 system, core 0 panics on EEPROM.commit. 
I can get around this by moving the commit outside the noInterrupts() <-> interrupts() but I am unsure what other impact this may have on the system.

kiwi64ajs, 2023-07-11:
I think the issue may be because, apparently, the noInterrupts() function doesn't do anything or it may not be defined for each/all ESP32 variants.

Can you try changing:

    noInterrupts() -> portDISABLE_INTERRUPTS()

and

    interrupts() -> portENABLE_INTERRUPTS()

In the code to see if it solves the problem. See:
  https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Arduino.h#L82-L86
  https://github.com/espressif/arduino-esp32/issues/7211

Maybe this issue is specific to the ESP32C3.

PontusO, 2023-07-11:
Sorry, that doesn't change anything. Here's the full panic listing:

16:13:11.836 -> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 
16:13:11.935 -> 
16:13:11.935 -> Core  0 register dump:
16:13:11.935 -> MEPC    : 0x4038a4ca  RA      : 0x4038a4b0  SP      : 0x3fc9ed80  GP      : 0x3fc8f400  
16:13:11.935 -> TP      : 0x3fc8c214  T0      : 0x4005890e  T1      : 0x403810fa  T2      : 0x00000000  
16:13:11.935 -> S0/FP   : 0x3fc90678  S1      : 0x0000⸮ESP-ROM:esp32c3-api1-20210207
16:13:18.358 -> Build:Feb  7 2021
16:13:18.392 -> rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)

It seems to me that the WDT timeout on CPU0 indicates that the EEPROM functions for the ESP32-C3 is depending on interrupts happening, otherwise it times out and the WDT fires. I'm not sure why we are seeing a POWERON reset code but I have not checked it more thoroughly.
For now I am keeping interrupts enabled during EEPROM.commit and making sure that the data has been written properly to the EEPROM/Flash area.

kiwi64ajs, 2023-07-11:
Well, maybe for the ESP32C3, we can exclude the "disable interrupts", EEPROM.commit, and "enable interrupts”.

It may even be that Espressif has taken care of this internally within their EEPROM library.

Can you try commenting out the interrupt-related code and see what happens?

Alex

PontusO, 2023-07-13:
I can confirm that it works when the interrupt related code is commented out.

kiwi64ajs, 2023-07-13:
Interesting...

It looks like they changed the internals of the EEPROM library in 2019 to use their Non-Volatile Storage (NVS) mechanism, so perhaps now disabling interrupts is no longer needed.

See: https://github.com/espressif/arduino-esp32/commit/619568db5bcc219091c653a5ced5e378b3a5643b

I’ve removed the interrupts code for the ESP32, but left it in for the ESP8266.

The changes are in version 2.0.16 which should become available soon

closed 2023-07-13 by kiwi64ajs