mrrwa/NmraDcc #76: Question: How to electrically connect DCC pin to tracks

abelouso, 2024-01-07:
Hello,
 I am new to DCC world, but do enjoy programming microcontrollers to control trains and track accessories. However, I do not know how to connect the signal pin of the Arduino to the tracks (e.g. where the ground and the pin input go and if there are any electrical components needed). I saw a schematic on-line that involved a resistor and and a diode, but I am not sure if that is universally applicable.

kiwi64ajs, 2024-01-07:
Hi,

Take a look at the schematic at: https://mrrwa.org/dcc-decoder-interface/

There are simpler ones if you don’t need to do DCC CV Programming like the one on this page here: https://www.arcomora.com/mardec/

You may want to just buy a IoTT DCC Interface PCB to get you started from here:

	https://myiott.org/index.php/iott-stick/communication-modules/dcc-interface

	https://www.tindie.com/products/tanner87661/dcc-interface-breakout-board-with-grove-port/

HTH

Alex Sheherd

DonWillsmer, 2024-10-28:
if you draw power from the DCC to drive the Arduino ( using a bridge rectifier an electrolytic capacitor  I have used 1000uF  a voltage regulator is optional )  then the feed to the DCC feed pin (usually pin 2)  can be straight from one of the DCC feeds through a resistor and a diode  ( I used 12k on the basis that 12v on the DCC could not push a current greater than 1mA , and a 1N914 diode with the cathode to the DCC.)  this works fine because it is referenced to the power supply.  This would exclude CV programming. 
If you use a separate power supply you would need to have a ground connection so I would avoid that with a 1n637 opto isolator.

Don

kiwi64ajs, 2024-10-29:
Hi Don,

How about we continue this thread offline? Email me at kiwi64ajs_at_gmail_dot_com

1) You can still program CVs in OPS Mode without the “Service Mode" CV ACK capability. I have a workaround for the lack of DCC Accessory Decoder Programming for some DCC Systems that impersonates a mobile decoder’s OPS Mode programming Address. 
OPS Mode programming will let you fine-tune the Servo Positions for the DCC Decoder live.

2) I suggest you do a combination of setting up the initial CV values in a table in the Arduino Sketch for the general starting points and then modifying it as required using OPS Mode CV programming "on the main" using a throttle or DecoderPro.

I can add some more code to your sketch as a starting point.

3) To make it easier, copy an existing DecoderPro DCC Accessory Decoder XML file and modify it to support your CV allocations. It's a relatively straightforward process. 

See here: https://github.com/JMRI/JMRI/blob/master/xml/decoders/MERG_DCCACC.xml for an example of a basic DCC Accessory Decoder XML File.

This one is for one of the MERG CBus Servo Decoders https://github.com/JMRI/JMRI/blob/master/xml/decoders/MERG_CBUS_canmio_svo.xml which has shared/linked sections of the overall decoder definition.

HTH

Alex

closed 2024-11-09 by kiwi64ajs