mrrwa/NmraDcc #77: Question: Can DCC reading be paused temporarily?

reduzent, 2024-02-10:
(It feels a bit awkward to ask questions in GitHub. If there is a better place, please let me know.)

I'm trying to build a turn-out decoder with an ATTiny85. I couldn't find a servo library that doesn't interfere with NmraDcc. So I decided to create the pulses in the code. However, while there is a DCC signal on the tracks, the servo becomes jittery. I guess that is because the interrupt has more priority than the delayMicroseconds() function. So, i thought about turning DCC reading off while the servo is moving. My idea basically is:

1. Read DCC
2. When a packet with the correct address is received, turn of DCC reading
3. Perform action that requires precise timing (like controlling a servo)
4. When action has finished, turn back on DCC reading.

I'm aware that I would miss packets during the time DCC reading is turned off, but I don't have a need to switch turn-outs in quick succession. 

So, is there a way to disable DCC reading termporarily? Or is there another way to get jitterless servo control?

This is the code I'm currently using, which disables the pulse after switching so that the servo stops trembling. It works, but it would be nice if I could move the servo slowly and steadily:

```
#include "NmraDcc.h"
NmraDcc  Dcc ;
DCC_MSG  Packet ;

const int adresse = 25; 

const int LED = 4;
const int relay = 3;
const int servo = 1;
int pos = 90;
unsigned long last_update;
unsigned long now;

void servo_pulse(int pin, int pos){
  int pulse = map(pos,0, 180, 850, 2050);                                  // in steps of 1 degree
  digitalWrite(pin, HIGH);
  delayMicroseconds(pulse);
  digitalWrite(pin, LOW);
  delay(20);
}

void notifyDccAccTurnoutOutput( uint16_t Addr, uint8_t Direction, uint8_t OutputPower )
{ 
  switch(Addr){
    case adresse:
      if(Direction < 1){
        digitalWrite(LED, LOW);
        digitalWrite(relay, LOW);
        pos = 65;
      }else{
        digitalWrite(LED, HIGH);
        digitalWrite(relay, HIGH);
        pos = 115;
      }
      last_update = millis();
      break;
      default:
      break;
  }
}

void setup()
{
  Dcc.pin(0, 2, 1);//Pin 2 is the interrupt pin on an UNO
  Dcc.init( MAN_ID_DIY, 10, CV29_ACCESSORY_DECODER | CV29_OUTPUT_ADDRESS_MODE, 0 );
  pinMode(LED, OUTPUT);
  pinMode(relay, OUTPUT);
  pinMode(servo, OUTPUT);
}

void loop()
{
  Dcc.process();
  now = millis();
  if ((now - last_update) < 65){
    servo_pulse(servo, pos);
  } 
}
```

reduzent, 2024-02-10:
Hm.. now while reading the code, I wonder whether it is possible to perform the NmraDcc.init() function in the main loop, setting it to a pin with no DCC signal, and then later back to pin 2.

reduzent, 2024-02-13:
It worked by detaching the interrupt while the servo is moving (see line 59):

```#include "NmraDcc.h"
#include <EEPROM.h>
NmraDcc  Dcc ;
DCC_MSG  Packet ;

const int adresse = 25; 

const int LED = 4;
const int relay = 3;
const int servo = 1;
const int pos_straight = 93;
const int pos_turn = 156;

bool state = LOW;
bool got_packet = false;

void servo_pulse(int pin, int pos){
  int pulse = map(pos,0, 180, 850, 2050);
  digitalWrite(pin, HIGH);
  delayMicroseconds(pulse);
  digitalWrite(pin, LOW);
  delay(21);
}

void notifyDccAccTurnoutOutput( uint16_t Addr, uint8_t Direction, uint8_t OutputPower )
{ 
  switch(Addr){
    case adresse:
      if(Direction < 1){
        state = LOW;
      }else{
        state = HIGH;
      }
      got_packet = true;
      EEPROM.write(adresse, state);
      break;
    default:
      break;
  }
}

void setup()
{
  Dcc.pin(0, 2, 1);
  Dcc.init( MAN_ID_DIY, 10, CV29_ACCESSORY_DECODER | CV29_OUTPUT_ADDRESS_MODE, 0 );
  pinMode(LED, OUTPUT);
  pinMode(relay, OUTPUT);
  pinMode(servo, OUTPUT);
  state = EEPROM.read(adresse);
  digitalWrite(LED, state);
  digitalWrite(relay, state);

}

void loop()
{
  Dcc.process();
  if ( got_packet ) {
    detachInterrupt(0);
    if (state) {
      for (int i = pos_straight; i <= pos_turn; i++) {
        servo_pulse(servo, i);
      }
      digitalWrite(LED, HIGH);
      digitalWrite(relay, HIGH);
      state = false;
    } else {
      for (int i = pos_turn; i >= pos_straight; i--) {
        servo_pulse(servo, i);
      }
      digitalWrite(LED, LOW);
      digitalWrite(relay, LOW);
      state = true;
    }
    got_packet = false;
    Dcc.pin(0,2,1);
    Dcc.init( MAN_ID_DIY, 10, CV29_ACCESSORY_DECODER | CV29_OUTPUT_ADDRESS_MODE, 0 );
  }
 
}
``

closed 2024-02-13 by reduzent

kiwi64ajs, 2024-02-16:
Hi Roman,

Sorry, I’ve been away on holiday, but I see you have figured out how to achieve what you needed to do.

We’d have to add pause()/resume() methods to do what you want, but your solution is probably ok.

I don’t think I’ve tried to drive servos from a Tiny85, but I’m a bit surprised you had issues as the AVRTinyCore has Tiny-specific support for servos, but obviously, it's still CPU load dependant…

Regards

Alex Shepherd