mrrwa/NmraDcc #78: Error compiling sketch NmraDccAccessoryDecoder_1

selkirk5934, 2024-03-06:
I am using the Arduino IDE level 1.8.19 and trying to compile sketch named NmraDccAccessoryDecoder_1 from the Accessory folder of the NmraDcc-master library.  The error  is 'DEFAULT_ACCESSORY_DECODER_ADDRESS' was not declared in this scope
I get the a similar error compiling sketch NmraDccMultifunctionDecoder_1  'DEFAULT_MULTIFUNCTION_DECODER_ADDRESS' was not declared in this scope
I have not made any changes to either sketch.  I have used them as distributed.

Can you please explain how to correct this error.

Thank you,
Dale Gloer

kiwi64ajs, 2024-03-07:
Hmmm…

I just tried building both of those NmraDcc examples on Arduino IDE 1.8.19 and 2.3.2 with the Arduino Uno board, and they compile ok for me.

What Board/Chipset are you compiling for, as there may be an unmet dependency or some other incompatibility that needs to be fixed if possible? 

Regards

Alex

selkirk5934, 2024-03-07:
Alex,

I was trying the compile for an Nano Old Bootloader.  I just tried compiling again for both a Uno and a Mega2560 and received the same error on all boards.

Nano chipset is Mega 32BF, Uno chipset ATMEGA328P, MEGA2560 chipset ATMEGA2560

I am at a loss to know how to continue.

Dale Gloer

kiwi64ajs, 2024-03-08:
You could try:
- create another empty Arduino Workspace folder
- download only the NmraDcc library
- build the examples
If all goes well, then there is probably something wrong with your normal Arduino Workspace.
Perhaps the NmraDcc library is in multiple places by mistake and you have a mishmash of versions.

Perhaps search your PC for the NmraDcc.h file and see if you find it in more than one folder and delete the ones that are not in the expected place.

selkirk5934, 2024-03-08:
I thought I replied to your last update with a close.  Anyway, I installed IDE 2.3.2 and the sketch compiles correctly.  So this can be closed.  
Thank you for your time.
Dale Gloer

closed 2024-03-08 by selkirk5934