mrrwa/NmraDcc #79: Write CV on PoM not work

gorillale, 2024-03-26:
Hi on POM Mode i cant write CV Values, no advancedAck will be called.
I use Arduino UNO as accessory decoder.

into SM mode ist ok to read write cv.
I not known what ist the problem?
have you disabled POM write?

With the other library AP_DCC it works to write CV on POM Mode.
For read i must be program andwer code.

Have anyone same problem, used with newset Library?
Regards
Andrew

kiwi64ajs, 2024-03-26:
Hi Andrew,

What type of decoder are you trying to write CVs to, and what sort of address?

i.e., is it a multi-function (Loco) decoder or an accessory decoder that uses one of several addressing modes?

Maybe send me the  AP_DCC example that works and the NmraDcc example that doesn’t work.

Alex

gorillale, 2024-03-27:
[NmraDccAccessoryDecoder_test.zip](https://github.com/mrrwa/NmraDcc/files/14769663/NmraDccAccessoryDecoder_test.zip)
Hi thank you for answere.
I have send test app and Electronic Sheet.
Same use with AP_DCC, other Code (that works).

kiwi64ajs, 2024-03-27:
Hi Andrew,

So you’re trying to write CVs in an Accessory Decoder with an address of 1001.

1) What type of DCC Command station are you using?

The issue is that the DCC Protocol has multiple different DCC Commands to perform a CV Write via PoM, and different DCC Systems do this differently.

Some DCC Systems don’t support Accessory Decoder CV PoM protocol at all. To provide a way to support PoM for Accessory Decoder on systems that don’t support it, I added a Multi-Function Decoder address that an Accessory Decoder would listen for Multi-Function Decoder PoM commands and do the CV Write that way.

To enable this put the MultiFunction DCC Address in the last parameter (OpsModeAddressBaseCV) of the Dcc.init() method. See below for details:

	void init (uint8_t ManufacturerId, uint8_t VersionId, uint8_t Flags, uint8_t OpsModeAddressBaseCV);

2) I see you have enabled DCC packet Printing using

	#define NOTIFY_DCC_MSG

Can you send me the raw DCC packets showing the Accessory Decoder CV PoM Commands, as that may help to debug this.

Regards

Alex Shepherd

m: +64-21-777764
e: ***@***.***

gorillale, 2024-03-27:
Hi
1)
Open DCC (arduino Mega), and BiDiB-IF2, 
But i wonder that AP_DCC works with these two Centrales and iTrain.

2)
Opening port
Port open
NMRA DCC Example 1
Init Done
66
//POM Write Command, no change 
notifyDccMsg: BB C8 EC 20 37 88 
notifyDccMsg: BB C8 EC 20 37 88 
notifyDccMsg: BB C8 EC 20 37 88 
notifyDccMsg: BB C8 EC 20 37 88 
// Read Command (not implemented)
notifyDccMsg: BB C8 E4 20 0 B7 
notifyDccMsg: BB C8 E4 20 0 B7 
notifyDccMsg: BB C8 E4 20 0 B7 
notifyDccMsg: BB C8 E4 20 0 B7 

Is OpsModeAddressBaseCV the Address of 1001?
regards Andrew

gorillale, 2024-06-26:
I have reprogrammed my code for AP_DCC Library and all Decoder work now fine.
I can Write CV but not Read (Timer Problems). Read is nice to have but not Need.
CV Write is ok with AP_DCC.

I have never found a soloution for NRMA Library to work to write CV values.
Thankyou for Help anyway.
Regards