mrrwa/NmraDcc #80: NMRADCC on Arduino R4 Wifi

SombreYellowBuffalo, 2024-06-16:
Aiming to migrate working sketches from UNO R3 to UNO R4 Wifi board - however including the NMRADCC library in an R4 sketch generates the below error (even if it is the only line in an otherwise new sketch). 
Any hope to have NMRADCC work on UNO R4 WiFi boards?

/Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp: In function 'void ExternalInterruptHandler()':
/Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:520:95: error: no matching function for call to 'attachInterrupt(uint8_t&, void (&)(), byte&)'
                     attachInterrupt (DccProcState.ExtIntNum, ExternalInterruptHandler, ISREdge);
                                                                                               ^
In file included from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Interrupts.h:8:0,
                 from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/ArduinoAPI.h:29,
                 from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/Arduino.h:4,
                 from /Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.h:52,
                 from /Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:49:
/Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Common.h:113:6: note: candidate: void attachInterrupt(pin_size_t, voidFuncPtr, PinStatus) <near match>
 void attachInterrupt(pin_size_t interruptNumber, voidFuncPtr callback, PinStatus mode);
      ^~~~~~~~~~~~~~~
/Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Common.h:113:6: note:   conversion of argument 3 would be ill-formed:
/Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:520:95: error: invalid conversion from 'byte {aka unsigned char}' to 'PinStatus' [-fpermissive]
                     attachInterrupt (DccProcState.ExtIntNum, ExternalInterruptHandler, ISREdge);
                                                                                               ^
In file included from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/ArduinoAPI.h:29:0,
                 from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/Arduino.h:4,
                 from /Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.h:52,
                 from /Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:49:
/Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Interrupts.h:38:27: note: candidate: template<class T> void arduino::attachInterrupt(pin_size_t, arduino::voidTemplateFuncPtrParam<T*>, PinStatus, T*)
 template<typename T> void attachInterrupt(pin_size_t interruptNum, voidTemplateFuncPtrParam<T*> userFunc, PinStatus mode, T* param) {
                           ^~~~~~~~~~~~~~~
/Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Interrupts.h:38:27: note:   template argument deduction/substitution failed:
/Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:520:95: note:   candidate expects 1 argument, 0 provided
                     attachInterrupt (DccProcState.ExtIntNum, ExternalInterruptHandler, ISREdge);
                                                                                               ^
In file included from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/ArduinoAPI.h:29:0,
                 from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/Arduino.h:4,
                 from /Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.h:52,
                 from /Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:49:
/Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Interrupts.h:21:27: note: candidate: template<class T> void arduino::attachInterrupt(pin_size_t, arduino::voidTemplateFuncPtrParam<T>, PinStatus, T&)
 template<typename T> void attachInterrupt(pin_size_t interruptNum, voidTemplateFuncPtrParam<T> userFunc, PinStatus mode, T& param) {
                           ^~~~~~~~~~~~~~~
/Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Interrupts.h:21:27: note:   template argument deduction/substitution failed:
/Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:520:95: note:   candidate expects 1 argument, 0 provided
                     attachInterrupt (DccProcState.ExtIntNum, ExternalInterruptHandler, ISREdge);
                                                                                               ^
/Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:577:87: error: no matching function for call to 'attachInterrupt(uint8_t&, void (&)(), byte&)'
             attachInterrupt (DccProcState.ExtIntNum, ExternalInterruptHandler, ISREdge);
                                                                                       ^
In file included from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Interrupts.h:8:0,
                 from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/ArduinoAPI.h:29,
                 from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/Arduino.h:4,
                 from /Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.h:52,
                 from /Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:49:
/Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Common.h:113:6: note: candidate: void attachInterrupt(pin_size_t, voidFuncPtr, PinStatus) <near match>
 void attachInterrupt(pin_size_t interruptNumber, voidFuncPtr callback, PinStatus mode);
      ^~~~~~~~~~~~~~~
/Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Common.h:113:6: note:   conversion of argument 3 would be ill-formed:
/Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:577:87: error: invalid conversion from 'byte {aka unsigned char}' to 'PinStatus' [-fpermissive]
             attachInterrupt (DccProcState.ExtIntNum, ExternalInterruptHandler, ISREdge);
                                                                                       ^
In file included from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/ArduinoAPI.h:29:0,
                 from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/Arduino.h:4,
                 from /Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.h:52,
                 from /Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:49:
/Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Interrupts.h:38:27: note: candidate: template<class T> void arduino::attachInterrupt(pin_size_t, arduino::voidTemplateFuncPtrParam<T*>, PinStatus, T*)
 template<typename T> void attachInterrupt(pin_size_t interruptNum, voidTemplateFuncPtrParam<T*> userFunc, PinStatus mode, T* param) {
                           ^~~~~~~~~~~~~~~
/Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Interrupts.h:38:27: note:   template argument deduction/substitution failed:
/Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:577:87: note:   candidate expects 1 argument, 0 provided
             attachInterrupt (DccProcState.ExtIntNum, ExternalInterruptHandler, ISREdge);
                                                                                       ^
In file included from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/ArduinoAPI.h:29:0,
                 from /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/Arduino.h:4,
                 from /Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.h:52,
[... 32 lines cut ...]
[... cut ...]
/Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp: In function 'uint8_t validCV(uint16_t, uint8_t)':
/Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.h:134:22: error: 'E2END' was not declared in this scope
     #define MAXCV    E2END          // the upper limit of the CV value currently defined to max memory.
                      ^
/Users/davidcooksey/Documents/Arduino/libraries/NmraDcc/NmraDcc.cpp:846:14: note: in expansion of macro 'MAXCV'
     if (CV > MAXCV)
              ^~~~~

exit status 1

Compilation error: exit status 1

kiwi64ajs, 2024-06-16:
> /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Interrupts.h:38:27: note: candidate: template void arduino::attachInterrupt(pin_size_t, arduino::voidTemplateFuncPtrParam<T*>, PinStatus, T*)
> template void attachInterrupt(pin_size_t interruptNum, voidTemplateFuncPtrParam<T*> userFunc, PinStatus mode, T* param) {
> /Users/davidcooksey/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/Interrupts.h:21:27: note: candidate: template void arduino::attachInterrupt(pin_size_t, arduino::voidTemplateFuncPtrParam, PinStatus, T&)
> template void attachInterrupt(pin_size_t interruptNum, voidTemplateFuncPtrParam userFunc, PinStatus mode, T& param) {

Hi SombreYellowBuffalo,

I’ve added a couple of MACRO definitions for the UNO R4 to the NmraDcc.cpp and NmraDcc.h files for you to try.

Download the most recent library code from here: https://github.com/mrrwa/NmraDcc/archive/refs/heads/master.zip

The code compiles ok, but I’ve not been able to test it on any hardware so please let me know if this works ok for you, so I can release the change.

Regards

Alex Shepherd

m: +64-21-777764
e: ***@***.***

SombreYellowBuffalo, 2024-06-16:
Hi Alex,

I imported the zip library into IDE and the simple (otherwise new sketch) verified without errors, so initially looks good.

I'll see if I can get the full script working now and let you know of any NMRADCC related issues.

Many thanks

David

SombreYellowBuffalo, 2024-06-17:
Everything is looking good now on Arduino R4 WiFi. Thanks for the quick response.
David

HuibMaaskant, 2024-08-23:
Today I ran into exactly the same issue with an Uno R4 Minima. Although the library should be updated (and I am using the latest version: 2.0.16), I am getting these errors. Any thoughts?

Thanks, Huib

kiwi64ajs, 2024-08-24:
I've committed the changes I had made for the Uno R4 Minima and bumped the version to 2.0.17.
It should be available in the Arduino Library Manager soon (maybe in an hour or two).

closed 2024-08-24 by kiwi64ajs

HuibMaaskant, 2024-08-24:
Thank you for fixing this. It works great! 

Grtz! Huib