mrrwa/NmraDcc #82: Question - Railcom support planned ?

chatelao, 2024-10-08:
I have a little question:

- Is there any RailCom support planned for the library?

kiwi64ajs, 2024-10-08:
No, not by me, at least... ;) 
It requires specialised electronic circuitry to facilitate the data transfer, which is beyond my capability.
Perhaps some brave and clever person will find a way to add it.

closed 2024-10-08 by kiwi64ajs

chatelao, 2024-10-15:
As soon as I will have the hardware issue solved, I will come back :-)

kiwi64ajs, 2024-10-15:
Wow, sounds exciting!!! ;)

chatelao, 2024-12-21:
Sneak preview of a OSHW 8x accessory decoder "coming soon" (aka 1 or 2 HW debug rounds left ahead).
=> I will be powered by any Seeed XIAO / Adafruit QT board you like - My favorit is still the XIAO RP2040

[image]