mrrwa/NmraDcc #83: Accessory Addresses not recognised

DonWillsmer, 2024-10-23:
I was getting addresses not being recognised. So set it to print received addresses to Serial . I found it was working fine until the address was anything over 253. This is using Output mode addressing i.e a simple number.

Entered 252   252 received over serial
             253   65533 received

anything above  253  gives either no response or a random number such as 929 when 2000 was entered or 429 for 901

This maybe a known limitation  of which  I was unaware or simply  a uint_8 which cannot take bigger addresses ( I tried 255 to see if that was the limit).

Priority is low I can  continue but it would obviously be a limitation for a very large system . My apologies if I have missed the limitation in documentation.

Don Willsmer

railnerd, 2024-10-23:
uint_8 can only hold a value from 0 to 255.-Dave

kiwi64ajs, 2024-10-23:
Hi Don,

Can you give me some more context around this, please?

> On 24 Oct 2024, at 10:25 AM, DonWillsmer ***@***.***> wrote:
> I was getting addresses not being recognised.

How are you getting this? What is telling you the address is not recognised?
> So set it to print received addresses to Serial . I found it was working fine until the address was anything over 253. This is using Output mode addressing i.e a simple number.
> 
I presume you’re running an Arduino Sketch and have debugging code that is telling you something.

 Can you send the sketch to me so I can see what you mean and what you are trying to achieve? 
> Entered 252 252 received over serial
> 253 65533 received
> 
Again, what does “Entered” mean?

Have you defined these address values in CVs in the Arduino sketch and forced them to be written to EEPROM, or are you programming them on a Programming track, or what?
> anything above 253 gives either no response or a random number such as 929 when 2000 was entered or 429 for 901
>
> This maybe a known limitation of which I was unaware or simply a uint_8 which cannot [...]
Yeah, sorry, but I’m finding it difficult to understand the context of your comments, and I can’t offer much help, until you can give more context as there are too many variables.

Regards

Alex

DonWillsmer, 2024-10-24:
Alex
 
thankyou very much for your quick response.
 
Some context I am developing an accessory decoder that will drive 
servos. I wanted them to be adjustable without needing to re-program or 
plug anything in. This would mean say in a club situation or installing 
in a friends layout I wouldn't need to be offering support just to 
adjust servos. 
I decided that having some set addresses to make the adjustments would 
be simpler so I chose 901 to increase the servo angle and 902 to 
decrease plus other to save the changes or change the Dcc address. I 
also recognised that  if there was multiple arduinos on a layout I had 
to be able to set only one to respond so each has a checkno that sets a 
flag to respond to those addresses. I chose 1780 at random in the tests.
Although it would operate the servos as I wanted it would not go into 
the setup mode. I thought it was my logic and kept trying to work out 
where I had gone wrong. Since it was receiving the correct address 5 or 
20 which I was using I didn't think that was the problem.
However in order to prove that the checkno address was being received 
and the flag was being set I tried 178 instead and that worked. So  I 
then printed to serial each address so entering 252 into the handset 
worked it showed 252 as the received address but entering 253 it printed 
a number outside the address range.
 
I didn't want to use CVs as there seems to be differences between Dcc 
systems and I was aiming for inexperienced users. As it is I have two 
systems one has separate buttons for On and Off on accessories the other 
just has  one button that toggles.
I dropped the adjustment codes to 201,202 etc and can successfully 
change the setting you just enter the Dcc code and it will slowly alter 
the end point. press to set Off and it will stop. I can also change the 
Dcc address. I haven't yet checked out the EEPROM save. I did allow some 
space for the details your library saves.
 
I attached the code this is work in progress so things might not work as 
they should.

#83)
 
 

 
<https://github.com/mrrwa/NmraDcc/issues/83#issuecomment-2433579267> ,

DonWillsmer, 2024-10-24:
Thank you for your email.
I am well aware of the size of a uint_8 which is why I suggested that 
could be an issue and the odd number returned was close to a rolled over 
uint_16  but the addresses  are valid. Alex Sheppard has been in touch 
and I have given him further details and my code.
 
Don

#83)
 
 

<https://github.com/mrrwa/NmraDcc/issues/83#issuecomment-2433522129> ,

kiwi64ajs, 2024-10-24:
Hi Don,

Your explanation of what you’re trying to achieve and the issues you described makes a lot of sense to me now… ;)

Your solution/approach seems interesting/novel but could be worth experimenting with to see how well it works. Managing the allocation of unique turnout addresses to tune each decoder will be a challenge/burden but doable.

However, your attachment didn’t survive the issues tracking system - maybe there is a setting somewhere to enable that, so just email it to me directly at ***@***.*** ***@***.***>

I’d still encourage you to structure your solution around CVs as I have a hack/mechanism to facilitate OPS Mode CV programming using a LocoMotive DCC Address to perform CV Writes “On the Main” as such. You can’t disable the CV programming in the library, but you can ignore it.

Also, people have been working on adding generalised support into JMRI for DCC Accessory Decoder OPS Mode Programming to support DCC Systems that don’t explicitly support OPS Mode Accessory CV programming BUT do provide a way to inject raw DCC packets onto the track - which gets around this problem. I had attempted to fix/add this several years ago but didn’t complete the work, but I think someone else may have taken up the challenge now - I need to check. 

Also, I’m way behind in getting some bugs/ambiguous behaviour sorted in the library for DCC Accessory Decoder CV Programming, as the DCC Specs describe several methods, but I only supported one.

I expect you’ve got the library in DCC Accessory Board Addressing mode, which allows for something like 512 (or maybe 511) board addresses with 4 turnouts on each = 2048 DCC Turnout Addresses. Whereas, you want to address the turnouts individually, which is fine but is a different mode, and it's not clearly documented… ;(

Once I see your code, I should be able to tweak it to behave as you hoped.

HTH

Regards

Alex Shepherd

DonWillsmer, 2024-10-25:
Alex 
 
thank you for your response. Unfortunately the system has disguised the 
email address in your text  so I cannot send the code. I will see if I 
can find another way.
 
Regarding your comments I find the CV set up confusing I have read 9.2.2 
Decoder CVs and did not find it easy to understand. As my needs are 
mostly related to the servos I am hoping to store them in EEPROM leaving 
space for the CVs. Also I have set it up so that when adjusting a servo 
end point  I can set the servo to increment at suitable intervals so a 
user can watch the turnout blades or signal arm moving and send the off 
command to stop the movement.  I need to store 10 bytes for each output 
( Dcc address, normal servo setting, reversed servo setting,  movement 
step, reserved in case ) these are loaded on start up. With 16 possible 
pins in a Nano that would be 160 byes plus 2 bytes for a check number. 
So not a lot of EEPROM. I just need to leave space for the CVs.
The only bit that has to be remember is the unique check no for each 
Arduino which used as a DCC address sets it into update more.  The code 
to perform the updates are common to all Arduinos as long as the Checkno 
is used only that Arduino will react to the update commands. Having 
changed the addresses used from 901,902 etc to 201,202 etc the code 
works as needed.  I flagged it up to you as I know some people place 
address for each station into different hundreds. So can easily use 
quite high numbers.
     
This is the bit in the code which flagged the incorrect numbers
 
void notifyDccAccTurnoutOutput(uint16_t receivedAddress, uint8_t 
direction, uint8_t outputPower) {
Serial.println(receivedAddress);  
 

As I said this works for addresses up to 252 but gives an incorrect 
address over that. I suspect that 252 addresses are enough for anyone so 
fixing this is probably not worthwhile but I did think it worth flagging 
up. I spent quite a bit of time thinking it was my code and didn't want 
others to do the same. 
 
I will append the code here to see if that gets through
 
Regards Don
 
 
#include <NmraDcc.h>
#include <EEPROM.h>
#define servopin 9
#define DCC_PIN 2

 
NmraDcc Dcc;
DCC_MSG Packet;
int checkno=178;       //used to indicate first values saved to EEPROM

 
int change=0;             // set the increment of the servo steps  
bool update=false;            // indicates the servo settings are being 
updated
bool updateAdd=false;   // indicates the DCC address is being updated.
bool dir=false;
int EEstart=198;
byte servono;

 
struct Servos {
int normal_pos=1000;
int reverse_pos=1900;
int current_pos;
int target;
int address=20;
[... 177 lines cut ...]
 
Dcc.process();

 
if (attarget==false) {
  //Serial.println(&quot;operating servo&quot;);
  Operate();
}

 
//if (reversed) reverse_pos+=change;  
//else normal_pos+=change;
//attarget=false;
if (change!=0) {
  Serial.println("Adjusting setting&quot;);
  Change();
}
}

#83)
 
 

 
<https://github.com/mrrwa/NmraDcc/issues/83#issuecomment-2436163164> ,

DonWillsmer, 2024-10-25:
Alex 
 
apologies I had started to change the code to handle multiple servos 
having been test with just one and picked up the wrong code this will 
compile and show the problem. It occurred to me I had caused a problem 
writing to EEPROM so I have changed the area of EEPROM but the problem 
is exactly the same so I dont think that could be the cause.
 
Regards Don
 
#include <NmraDcc.h>
#include <EEPROM.h>
#define servopin 9
#define DCC_PIN 2

 
NmraDcc Dcc;
DCC_MSG Packet;
int checkno=178;       //used to indicate first values saved to EEPROM

 
int change=0;             // set the increment of the servo steps  
bool update=false;            // indicates the servo settings are being 
updated
bool updateAdd=false;   // indicates the DCC address is being updated.
bool dir=false;
int EEstart=598;

 
//struct Servos {}
int normal_pos=1000;
int reverse_pos=1900;
int current_pos;
int target;
int address=20;
unsigned long lastpulse;
unsigned long lastchange;
int step=10;
bool reversed;
bool attarget;

 
void Operate() {
  if (micros()-lastpulse>19000) {
  lastpulse=micros();
 
  if (current_pos<target) {
    current_pos+=step;
    if (current_pos>target) current_pos=target;
    }
  else if (current_pos>target) {
    current_pos-=step;
    if (current_pos<target) current_pos=target;
    }
 
    noInterrupts();
    digitalWrite(servopin,HIGH);
    delayMicroseconds(current_pos);
    digitalWrite(servopin, LOW);
    interrupts();
     if (current_pos==target) attarget=true;
  }
}

 
void Change() {
  if (millis()-lastchange>200) {
   
    if (reversed) {
      reverse_pos+=change;
      current_pos=reverse_pos;
    }
    else {
      normal_pos+=change;
      current_pos=normal_pos;
    }  
    noInterrupts();
    digitalWrite(servopin, HIGH);
    delayMicroseconds(current_pos);
    digitalWrite(servopin,LOW);
    interrupts();
    lastchange=millis();
   }
  }

 
void Save() {
  EEPROM.put(EEstart+2, normal_pos);
  EEPROM.put(EEstart+4, reverse_pos);
  EEPROM.put(EEstart+6, address);
  EEPROM.put(EEstart+8, step);
}

 
void notifyDccAccTurnoutOutput(uint16_t receivedAddress, uint8_t 
direction, uint8_t outputPower) {
Serial.println(receivedAddress);  
  if (receivedAddress==address) {
    if (direction) {
      target=reverse_pos;
      reversed=true;
    }
    else {
      target=normal_pos;
      reversed=false;
    }
    attarget=false;
  }
  else  if (receivedAddress==checkno) {
    update=true;
    Serial.println(update);
    }

 
  if (update==true) {

 
 /* if (receivedAddress==901) {
    if (direction) change=step;
    else change=0;

 
    }
  if (receivedAddress==902) {
    if (direction) {
      change=-step;
    }
    else {
      change=0;
    }
  }*/

 
   
    switch (receivedAddress) {
      case 201:
      if (direction)  change=step;
      else change=0;
      break;
      case 202:  
      if (direction) change=-step;
      else change=0;
      break;
      case 204:
      Save();
      break;
      case 203:
      updateAdd=true;
      break;
      case 205:
      update=false;
      break;
      default:
      if (updateAdd==true) {
        address=receivedAddress;
        updateAdd=false;
      }
     // else
     // if receivedAddress=
     
     
      }  
    }
}

 
void setup() {
  // put your setup code here, to run once

 
byte test=EEPROM.read(0);
if (test!= checkno) {
  Save();

 
  EEPROM.write(EEstart, checkno);
}
else {
  EEPROM.get(EEstart +2, normal_pos);
  EEPROM.get(EEstart +4, reverse_pos);
  EEPROM.get(EEstart +6, address);
  EEPROM.get(EEstart +8, step);
}

 
Dcc.pin(0, 2, 1);
Dcc.init(MAN_ID_DIY, 10, CV29_ACCESSORY_DECODER | 
CV29_OUTPUT_ADDRESS_MODE, 0);

 
Serial.begin(115200);
delay(5000);
Serial.println("Serial started&quot;);

 
current_pos=normal_pos;
target=normal_pos;
noInterrupts();
pinMode(servopin, OUTPUT);
digitalWrite(servopin, HIGH);
delayMicroseconds(current_pos);
digitalWrite(servopin,LOW);
interrupts();
attarget=true;
}

 
void loop() {
  // put your main code here, to run repeatedly:

 
Dcc.process();

 
if (attarget==false) {
  //Serial.println(&quot;operating servo&quot;);
  Operate();
}

 
//if (reversed) reverse_pos+=change;  
//else normal_pos+=change;
//attarget=false;
if (change!=0) {
  Serial.println("Adjusting setting&quot;);
  Change();
}
}

#83)
 
 

 
<https://github.com/mrrwa/NmraDcc/issues/83#issuecomment-2436163164> ,

kiwi64ajs, 2024-10-27:
Hi Don,
I've tweaked a few things in your code around what you were doing with the EEPROM, as the Library uses the EEPROM for CV Values as well, so I moved your usage of EEPROM offset to: CV_MANUFACTURER_START.
But, apart from that, I don't really see anything obvious as to why you get the output you're seeing as the code seems to work ok for me.
I can enter the Turnout Addresses 1..2044 (the first and last valid address), into my throttle and control the turnout position, and the call-back function is called with the correct values.

For testing, I'm using version 2.0.17 of the library on an Arduino Uno, a TCS CS-105 Command Station and a UWT-100 throttle.
Here is the modified code I used for the test.

BTW I added some of the CV Definitions in case you wanted to use those, but they're not actually doing anything to fix the issue.

```
#include <NmraDcc.h>
#include <EEPROM.h>
#define servopin 9
#define DCC_PIN 2

NmraDcc Dcc;
DCC_MSG Packet;

struct CVPair
{
  uint16_t  CV;
  uint8_t   Value;
};

CVPair FactoryDefaultCVs [] =
{
  {CV_ACCESSORY_DECODER_ADDRESS_LSB, DEFAULT_ACCESSORY_DECODER_ADDRESS & 0xFF},
  {CV_ACCESSORY_DECODER_ADDRESS_MSB, DEFAULT_ACCESSORY_DECODER_ADDRESS >> 8},
};

uint8_t FactoryDefaultCVIndex = 0;

int checkno = 178;  //used to indicate first values saved to EEPROM

int change = 0;          // set the increment of the servo steps
bool update = false;     // indicates the servo settings are being updated
bool updateAdd = false;  // indicates the DCC address is being updated.
bool dir = false;
// int EEstart = 598;
int EEstart = CV_MANUFACTURER_START;

//struct Servos {}
int normal_pos = 1000;
int reverse_pos = 1900;
int current_pos;
int target;
int address = 20;
unsigned long lastpulse;
unsigned long lastchange;
int step = 10;
bool reversed;
bool attarget;

void Operate() {
  if (micros() - lastpulse > 19000) {
    lastpulse = micros();

    if (current_pos < target) {
      current_pos += step;
      if (current_pos > target) current_pos = target;
    } else if (current_pos > target) {
      current_pos -= step;
      if (current_pos < target) current_pos = target;
    }

    noInterrupts();
    digitalWrite(servopin, HIGH);
    delayMicroseconds(current_pos);
    digitalWrite(servopin, LOW);
[... 144 lines cut ...]
  //attarget=false;
  if (change != 0) {
    Serial.println("Adjusting setting");
    Change();
  }

  if( FactoryDefaultCVIndex && Dcc.isSetCVReady())
  {
    FactoryDefaultCVIndex--; // Decrement first as initially it is the size of the array
    uint16_t cv = FactoryDefaultCVs[FactoryDefaultCVIndex].CV;
    uint8_t val = FactoryDefaultCVs[FactoryDefaultCVIndex].Value;
#ifdef DEBUG_MSG
    Serial.print("loop: Write Default CV: "); Serial.print(cv,DEC); Serial.print(" Value: "); Serial.println(val,DEC);
#endif     
    Dcc.setCV( cv, val );
  }
}

void notifyCVResetFactoryDefault()
{
  // Make FactoryDefaultCVIndex non-zero and equal to num CV's to be reset 
  // to flag to the loop() function that a reset to Factory Defaults needs to be done
  FactoryDefaultCVIndex = sizeof(FactoryDefaultCVs)/sizeof(CVPair);
};
```

DonWillsmer, 2024-10-27:
This issue has been resolved the problem did not lie within the library but was a problem with the DCC system used. I will report the problem to MERG as it was built from their kit.
I have thanked Alex for his excellent support.

DonWillsmer, 2024-10-27:
Alex
 
I must apologise. I have been using a MERG CANCMD2 for testing and it 
still gave me the problem. As you have tested the code and found it to 
work I  have now tried it with a DCCEX system using a UWT50 throttle and 
can  use  accessory addresses 1 to 2004. I do apologise for taking up 
your time. I had thought the problem had been with my code or I had 
missed something. I didn't expect the issue lay with the CANCMD. I must 
say the prompt response and quick checking on your part is very kind. I 
will now report the issue to MERG. 
I also appreciate your tweaking of the code. I assume that moving the 
EEPROM to the Manufacturers area will avoid any problems should the 
other EEPROM areas become used.
 
I have also closed the issue on Github.
Don

#83)
 
 

   Serial.print(&quot;\nnotifyDccAccTurnoutOutput: Address: &quot;);
   Serial.print(&quot; Dir: &quot;);

   Serial.println(&quot;Serial started&quot;);

     Serial.println(&quot;Adjusting setting&quot;);
     Serial.print(&quot;loop: Write Default CV: &quot;); 
Serial.print(cv,DEC); Serial.print(&quot; Value: &quot;); 
<https://github.com/mrrwa/NmraDcc/issues/83#issuecomment-2439935304> ,

closed 2024-10-27 by DonWillsmer

kiwi64ajs, 2024-10-27:
Hi Don,

It’s good to resolve this as some things can be a bit baffling.

Sadly, the NMRA DCC Specs for DCC Accessory Control has several ambiguous definitions relating to Addressing, resulting in multiple similar but not the same implementations, which isn’t helpful… ;(

If your hunch is correct, the MERG guys have some remedial work to do on their system, but as you can tell, it's a bit of a minefield lacking clarity.

Regards

Alex Shepherd

kiwi64ajs, 2024-10-27:
Hi Don,

I’ve also committed some more changes after an email exchange with an ESP32 developer and removed the need for custom ESP32 macros, so hopefully, the ESP32 support will be more generic across the ESP32 range of MCUs.

Regards

Alex Shepherd