mrrwa/NmraDcc #86: Question - Any reason for hugh memory differences?

xDuinoRails, 2024-11-19:
Good evening everybody,

Just out of curiosity I compared the size of the same Arduino Sketch (NmraDccAcessoryDecocer). The results are surprisng:

| Mikrocontroller | FLASH | RAM |
| --- | --- | --- |
| Arduino UNO | 60854 | 388 |
| STM32F3 | 30120 | 3656 |
| RP2040 | 66524 | 9996 |
| RISC-V (ESP32-C3) | 294574 | 12556 |

kiwi64ajs, 2024-11-19:
I expect everything except the Arduino UNO will have a full version of FreeRTOS, and who knows what else (maybe a full Wi-Fi TCP/IP Stack) will be included by default.

You’d have to dump out the memory map to better understand what is in there…

It is probably the same reason why Turbo Pascal used to fit in 39kb of RAM on a Z80 back in the day, and now, who knows how many GB or RAM it needs to run on a Windows PC…

There isn’t much incentive to optimise the RAM/FLASH resource usage on these newer chips, so it's probably not got much attention.

Alex

xDuinoRails, 2024-11-20:
Thank you, if I find out more, I will post it here.

closed 2025-02-05 by kiwi64ajs