mrrwa/NmraDcc #87: Help request

Petej42, 2024-11-23:
Is anyone available to take a look at some arduino code using the  nmradcc library.
I am trying to put something together that allows me to control 17 servo controlled turnouts. The turnout attributes are stored as objects in an array, however when I call dcc.process, the last element in the turnout array gets corrupted and I cannot determine why, even with the code stripped back yto a minimum

Petej42, 2024-11-23:
[TurnoutControl_bare.ino.txt](https://github.com/user-attachments/files/17882642/TurnoutControl_bare.ino.txt)

kiwi64ajs, 2024-11-24:
I think your Turnout Array is one element short. Try changing this line:

	Turnout to[noTurnouts-1];

to:

	Turnout to[noTurnouts];

HTH

Alex Shepherd

mijnmodelbaan, 2024-11-24:
I think Alex is right and if not, what happens if you try 16 servos?

BR/Willem.

closed 2024-11-24 by Petej42