mrrwa/NmraDcc #90: Using Preferences instead of depreciated EEPROM library for esp32 hardware?

mikerinderknecht, 2025-01-28:
Hello, 
Would it be possible to use the Preferences library instead of depreciated EEPROM library for esp32 hardware?
Best,
Mike

kiwi64ajs, 2025-01-28:
Hi Mike,

> On 28 Jan 2025, at 8:04 PM, mikerinderknecht ***@***.***> wrote:
> Hello,
> Would it be possible to use the Preferences library instead of depreciated EEPROM library for esp32 hardware?
> 
Technically, we could, but you have to ask, “Why?” as I don’t expect the EEPROM library to disappear anytime soon, as an awful lot of code uses it, and the ESP32 Preferences library is specific to the ESP32 range and not available across all Arduino compatible devices. Also, after a quick read of the ESP32 Preferences API Docs, it is not clear to me when the Preferences are written to the NVS, and there look to be instances where the NVS needs to be erased and reformatted, so its use is not without issues and certainly not magic or a silver bullet… ;)

However, my thoughts are more toward creating a pluggable “Storage” class with a default implementation that uses the EEPROM library. This would allow the programmer to develop an alternative implementation that uses a non-volatile storage mechanism, such as the ESP32 Preferences, external I2C EEPROM or FRAM chips, etc.

That would also let us read all the values into RAM to speed up access, as some storage systems can be pretty slow. Also, writes could be cached and written after a delay to group multiple changes and write them to a page of FLASH memory in a single operation.

What do you think or propose?

Regards

Alex Shepherd

mikerinderknecht, 2025-01-29:
Hi Alex, I think you are spot on. Ideally it would be an agnostic implementation. I was also thinking whether to use an external EEPROM for ESP32 Devkit v1, but saw that it does not seem to be supported yet, so I started to think about the preferences library. Do you have any concrete plans on making the library agnostic with a pluggable storage class?
Best, Mike