mrrwa/NmraDcc #101: Question use DFR0282 (Arduino Leonardo)

gorillale, 2025-03-13:
Hi
I not known how to work the DFrobot 0282 (alias Beetle alias Arduino Leonardo) with the NMRA DCC.
I not known with Port i sould be use.
I have testet with D10 bit the Beetle have only an PCINT6 on this port.
Port 2 i cant use.

Is this NMRA-DCC Library to support the Beetle?

Thank you and regards.
andrew

kiwi64ajs, 2025-03-13:
Hi Andrew,

> 
> 
> gorillale
>  created an issue 
> (mrrwa/NmraDcc#101)
> Hi
> I not known how to work the DFrobot 0282 (alias Beetle alias Arduino Leonardo) with the NMRA DCC.
> I not known with Port i sould be use.
> I have testet with D10 bit the Beetle have only an PCINT6 on this port.
> 
> 
> gorillale
>  created an issue 
> (mrrwa/NmraDcc#101)
> Hi
> I not known how to work the DFrobot 0282 (alias Beetle alias Arduino Leonardo) with the NMRA DCC.
> I not known with Port i sould be use.
> I have testet with D10 bit the Beetle have only an PCINT6 on this port.
The library doesn’t support the PinChange interrupts - yet… ;) 
> Port 2 i cant use.
> 
> Is this NMRA-DCC Library to support the Beetle?
Hmmm… I’ve never used one of these “ Beetle” boards, but I think I have a Leonardo board somewhere at home I could try if necessary.

However suggest you try connecting the DCC Signal to the D0 (Rx) input and select external hardware INT 2

You’ll need to change the hardware interrupt and arduino pin number in the pin() method - try this:

	Dcc.pin(2, 0, 1);

HTH

Alex

gorillale, 2025-03-14:
Hi thank you for this qick answer.
Yes this sould be work, but the rx and tx has only 2 solder pins.
I need the plug, D9, D10, D11, A0 A1, A2 and Power.
I try with NESTED IQR and PCINT6 but not work.
For this Chip AVR32U4 i think, must be add other Codes to work with PCINT

Then i will test the next board d1 mini Pro...
regards
andrew

kiwi64ajs, 2025-03-14:
Hi Andrew,

> 
> 
> gorillale
>  left a comment 
> (mrrwa/NmraDcc#101)
> Hi thank you for this qick answer.
> Yes this sould be work, but the rx and tx has only 2 solder pins.
> I need the plug, D9, D10, D11, A0 A1, A2 and Power.
> I try with NESTED IQR and PCINT6 but not work.
> For this Chip AVR32U4 i think, must be add other Codes to work with PCINT
> 
> 
> gorillale
>  left a comment 
> (mrrwa/NmraDcc#101)
> Hi thank you for this qick answer.
> Yes this sould be work, but the rx and tx has only 2 solder pins.
> I need the plug, D9, D10, D11, A0 A1, A2 and Power.
> I try with NESTED IQR and PCINT6 but not work.
> For this Chip AVR32U4 i think, must be add other Codes to work with PCINT
1) You can only use External Hardware Interrupts, not PinChange Interrupts, as they use a different mechanism and are slower.

2) Looking at the Leonardo Schematic here: https://docs.arduino.cc/resources/schematics/A000057-schematics.pdf

The Hardware Interrupts are on these pins:
- INT0 -> PD0, SCL -> D3
- INT1 -> PD1, SDA -> D2
- INT2 -> PD2, RXD -> D0
- INT3 -> PD3, TXD -> D1

3) Looking at the Beatle IO Port Mapping here: https://wiki.dfrobot.com/Beetle_SKU_DFR0282#target_2

You should be able to use any of the Pins D0, D1, D2, D3

4) Lookign at the Beatle documentation the D0, D1, D2, D3 pins are available as solder pads on the underside of the Beatle PCB.

See: https://dfimg.dfrobot.com/5d09937f437c63049d98b469/wiki/657225f47e05a1f923c35c2c58f57528.png there are pads marked SDA, SCL, RX, TX in the centre of the PCB which you should be able to use.

4) You can use another variant of the Dcc.Pins() method, which should make the mapping easier

	Dcc.pin(DCC_PIN, 1);

Where DCC_PIN can be set to DO, D1, D2, D3 to match the Arduino pin you are using 

HTH

Alex

gorillale, 2025-03-14:
Hi, thank you
Yes the 2 and 3 i not need, then i can use this.
RX TS i used on Serial.

regards
Andrew

closed 2025-03-14 by gorillale