mrrwa/NmraDcc #102: Changing PWM frequency disables DCC communication

marko-pi, 2025-05-05:
Hello, this is more of an upgrade request than a issue.

I have developed a mini locomotive decoder based on NmraDCC (https://github.com/marko-pi/MPED-DIY). It uses PWM to control the motor speed and works fine, except for the annoying noise. The reason for this is that the default PWM frequency is 976.56 Hz.

Dr Geoff Bunza suggested changing the PWM frequency with the command

TCCR0B = TCCR0B & B11111000 | B00000001; // for a PWM frequency of 62500.00 Hz
TCCR0B = TCCR0B & B11111000 | B00000010; // for a PWM frequency of 7812.50 Hz
TCCR0B = TCCR0B & B11111000 | B00000011; // for a PWM frequency of 976.56 Hz (the DEFAULT)
TCCR0B = TCCR0B & B11111000 | B00000100; // for a PWM frequency of 244.14 Hz
TCCR0B = TCCR0B & B11111000 | B00000101; // for a PWM frequency of 61.04 Hz

However, as soon as the PWM frequency has been changed, DCC communication no longer works.

I would like to know if there is a way to adapt the library so that it also works for other PWM frequencies?

kiwi64ajs, 2025-05-05:
Hi marko-pi,

> On 5 May 2025, at 7:11 PM, marko-pi ***@***.***> wrote:
> 
> 
> marko-pi
>  created an issue 
> (mrrwa/NmraDcc#102)
>  <https://github.com/mrrwa/NmraDcc/issues/102>
> Hello, this is more of an upgrade request than a issue.
> 
> I have developed a mini locomotive decoder based on NmraDCC (https://github.com/marko-pi/MPED-DIY). It uses PWM to control the motor speed and works fine, except for the annoying noise. The reason for this is that the default PWM frequency is 976.56 Hz.
> 
> Dr Geoff Bunza suggested changing the PWM frequency with the command
> 
> TCCR0B = TCCR0B & B11111000 | B00000001; // for a PWM frequency of 62500.00 Hz
> TCCR0B = TCCR0B & B11111000 | B00000010; // for a PWM frequency of 7812.50 Hz
> TCCR0B = TCCR0B & B11111000 | B00000011; // for a PWM frequency of 976.56 Hz (the DEFAULT)
> TCCR0B = TCCR0B & B11111000 | B00000100; // for a PWM frequency of 244.14 Hz
> TCCR0B = TCCR0B & B11111000 | B00000101; // for a PWM frequency of 61.04 Hz
> 
> However, as soon as the PWM frequency has been changed, DCC communication no longer works.
> 
> I would like to know if there is a way to adapt the library so that it also works for other PWM frequencies?
> 
I don’t think there is a simple fix for what you’re wanting as the problem is that those commands change the Timer0, which the Arduino core uses for the timebase for micros() and millis(), and the NmraDcc library uses for all its internal timings.

I suggest you use the PWM outputs that are associated with Timer1 (D9, D10) or Timer2 (D3, D11), as you should be able to change the PWM frequency of those Timers/pins without altering the Arduino core timings.

HTH

Regards

Alex Shepherd

m: +64-21-777764
e: ***@***.***

marko-pi, 2025-05-05:
> I don’t think there is a simple fix for what you’re wanting as the problem is that those commands change the Timer0, which the Arduino core uses for the timebase for micros() and millis(), and the [...]

Yes, I have found that out in the meantime.

The problem is that I should redesign the PCBs, so I was hoping to find a patch. Perhaps manually changing the library by dividing the output of `millis()` by `64` for `62.5kHz` would solve the problem? Of course, I would also need to add an overflow check.

kiwi64ajs, 2025-05-05:
> marko-pi
>  left a comment
> (mrrwa/NmraDcc#102)
> Of course, the question arises as to whether the DIY locomotive decoder is worth making. I almost forgot about my project, but there is a Kiwi who wants to build my decoder and has already ordered the PCBs.
>
> Maybe the idea is to patch my design at least for testing purposes, and if it turns out to work well, develop a new design with different PWM pins.

I’ve not delved into the Arduino Uno Core Timers, but perhaps you can tweak the core functions that use Timer0

This may be useful: https://arduino.stackexchange.com/questions/13518/adjust-time-calculation-after-timer0-frequency-change

HTH

Alex

marko-pi, 2025-05-05:
Replacing `micros()` with `micros()/64` worked for `62.5kHz`... No more sqeaking. I could also get `7.8kHz` and `244Hz` to work, but not `61Hz`. I suspect that `micros()*16` was too coarse.

closed 2025-05-05 by marko-pi