mrrwa/NmraDcc #107: Loco and Accessory mode

Jacques44, 2025-09-08:
I'm using the NmraDcc library on a custom Nano-based PCB to drive different types of accessories (lights, servos, etc.).

For my application, I would like to receive callbacks from both loco decoder messages (via notifyDccSpeed) and accessory decoder messages (via notifyDccAccState) in the same sketch.

Currently, the library seems to support only one mode at a time (either loco or accessory).

As a workaround, I'm trying to reimplement accessory decoding inside notifyDccMsg, but this is quite complicated and duplicates part of the library logic.

Would it be possible to extend the library to support both types of callbacks simultaneously, or provide guidance on the recommended approach?

Jacques44, 2025-09-16:
[image]

kiwi64ajs, 2025-09-16:
Excellent!
Great that you found a way to achieve your desired outcome. What does the system in the photo do?

Jacques44, 2025-09-27:
I’ve designed a custom HO-scale diorama featuring a variety of advanced effects, including custom lighting, a [chain-driven road system](https://makerworld.com/en/models/561127-ho-road-chain-car-system-chain-system#profileId-480437), audio, and [custom signals](https://makerworld.com/en/models/682526-mrj-ho-scale-db-signals-era-iii-3x-faces#profileId-611046). My goal is to control all these elements using DCC (Digital Command Control).
To achieve this, I’ve developed a custom PCB based on an Arduino Nano. The PCB drives serial servos, [custom lamps](https://makerworld.com/en/models/588812-mrj-old-street-lamp-ho-scale-requires-0-2mm#profileId-510102) with persistence-of-vision (POV) effects, audio modules, OLED displays, I2C devices, and signals using charlieplexing. The lighting effects, such as gas lamps, beacons, storms, fires, filament lamps, traffic lights, oil lamps, and neon signs, are simulated using digital outputs with coroutine-based PWM emulation for smooth transitions. All components, including the models, were designed and 3D-printed by me.
The PCB uses a 6N137 optocoupler to receive DCC commands, an LM7805 for power regulation, and a comprehensive set of resistors and screw terminals to connect the various devices.
Currently, I’m facing significant memory limitations with the Arduino Nano. To address this, I plan to redesign the system using ESP32 microcontrollers for improved performance and memory capacity.